maximkulkin/esp-homekit-demo #2: temperature sensor demo question

faronov, 2017-12-10:
I think it may be better to use ESP-OPEN-RTOS. for base
it has more sensor support 
but what type of temp sensor this demo is supported?

renssies, 2017-12-10:
Currently the demo just sends a random number between 0 and 100 I believe. It doesnt actually use a sensor.

The LED demo uses the built in LED of most ESPs

maximkulkin, 2017-12-11:
Exactly. I did not want to mess with any external libraries or in any other way complicate examples. Temperature sensor example was supposed to show how to handle values that are updated on their own, how to declare and use characteristics.

Although I might add real DHT11 implementation sometime later.

maximkulkin, 2017-12-11:
Here you go guys! I've updated temperature_sensor example with real DHT11 integration code plus it now also exposes humidity information (because why not). Enjoy!

closed 2017-12-11 by maximkulkin

medmile, 2017-12-11:
I have error “can not read data from sensor” error in bloc “B” (DHT11). On ESP8266_RTOS_SDK sensor work good. Am I need to use resistor?

maximkulkin, 2017-12-11:
That is probably because you've specified incorrect GPIO pin number. Pin numberss might be different from Espressif SDK (myself had to do a couple of trial and error attempts before settling down correct one).

medmile, 2017-12-12:
I test nodemcu v3. Pin №4 (D2). Led example work on this pin.

maximkulkin, 2017-12-12:
Here is how I tested: I opened NodeMCU pinout 
[image], set SENSOR_PIN=4 in Makefile and connect DHT11 data output to NodeMCU's D2 (GPIO4). Although this method did not work for me for combination GPIO16 and D0.

maximkulkin, 2017-12-12:
That is strange because this is exactly the set up that worked for me. Could you provide your wiring schematics.

medmile, 2017-12-12:
[image]
[image]

maximkulkin, 2017-12-12:
Not sure what kind of break out you've got, but on mine (which looks exactly like this) data pin is in center while on your layout it's on the left. Are you sure you've connected everything properly?

maximkulkin, 2017-12-12:
Ok, the image of DHT is very blury, but it looks like your pin layout is correct. What I've noticed is: my breakout has embedded pull-up resistor (along with capacitor), so maybe you need to change [line 60](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/temperature_sensor/temperature_sensor.c#L60) to enable pull-up on ESP8266 side:
`gpio_set_pullup(SENSOR_PIN, true, false);`

medmile, 2017-12-12:
I use this module. This work with ESP8266_RTOS_SDK...
[image]

maximkulkin, 2017-12-12:
Well, I'm out of ideas. I suggest you ask help in [esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos) project. Try starting with their simple [dht_sensor](https://github.com/SuperHouse/esp-open-rtos/tree/master/examples/dht_sensor) example and make it work.

medmile, 2017-12-12:
I noticed that there is a 10kom resistor on my module. Could this be the reason? Do you use a resistor?

medmile, 2017-12-13:
I found mistake!
In dht.c first argument "dht_sensor_type_t":
bool dht_read_float_data(dht_sensor_type_t sensor_type, uint8_t pin, float *humidity, float *temperature)
{
    int16_t i_humidity, i_temp;

    if (dht_read_data(sensor_type, pin, &i_humidity, &i_temp)) {
        *humidity = (float)i_humidity / 10;
        *temperature = (float)i_temp / 10;
        return true;
    }
    return false;
}
but you use first argument "SENSOR_PIN":
bool success = dht_read_float_data(
            SENSOR_PIN, DHT_TYPE_DHT11,
            &humidity_value, &temperature_value
        );

maximkulkin, 2017-12-13:
Oh! Thank you for reporting that. Fixed.

faronov, 2017-12-28:
I see another strange behavior
It shows temperature and humidity, but continuously I get error
`HomeKit: [Client 5] Error reading data (code 128)`
It looks like it ask too many times for getting data
after some time it crashed

maccoylton, 2017-12-28:
faronov, your not alone, I'm getting that error as well

maximkulkin, 2017-12-29:
Pushed change to force disconnect client in case reading from socket results in an error.

faronov, 2017-12-29:
As I see it fail after it shows IP
I increase vtask time , but it didn't help
```
>>> mdns_recv: pbuf too big
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
Temperature sensor identify
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
Temperature sensor identify
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: Got new client connection: 6
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: [Client 6] Get Accessories
ip:192.168.11.23,mask:255.255.255.0,gw:192.168.11.1
!!! HomeKit: [Client 5] Error reading data (code 104)
!!! HomeKit: [Client 5] Error reading data (code 104)
```

faronov, 2017-12-29:
Next run
same result
dhcp client start...
```
ip:192.168.11.23,mask:255.255.255.0,gw:192.168.11.1
>>> HomeKit: Got IP, starting
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
Temperature sensor identify
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
Temperature sensor identify
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: Got new client connection: 6
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with 1F4D9525-9246-430F-B309-538363BA9646
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: [Client 6] Get Accessories
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
ip:192.168.11.23,mask:255.255.255.0,gw:192.168.11.1
Couldnt read data from sensor

```

maximkulkin, 2018-01-03:
`Couldnt read data from sensor` seem to be irrelevant to HomeKit.

Regarding `!!! HomeKit: [Client 5] Error reading data (code 104)`: this should be fixed in latest code, server will drop clients with (unknown) socket errors. I found another problem: sometimes it does not detect clients being gone and eventually accumulates maximum number of clients (8), refusing to accept new ones. I'm doing long term testing of a fix for this one and will push soon.

BTW, how do you trigger "Identify" multiple times without repairing? I couldn't find a button in Home.app to do that.

faronov, 2018-01-03:
I think one identity from thermometer
and second from humidity

HomeACcessoryKid, 2018-01-03:
Identify is accessible if you use the Eve App from ElGato

orisraeli, 2018-08-27:
could this work with DHT22 sensor?

maximkulkin, 2018-08-27:
Yes, the library that is used there supports both DHT11 and DHT22, you just need to pass it which version of sensor you have.

LiangyongFShen, 2019-02-25:
@faronov Have you solved your problem? I have the same problem, except the separate pairing switch, other devices cannot be paired。