maximkulkin/esp-homekit-demo #3: unpair and repair

medmile, 2017-12-11:
Hello.

I build example successfully, make pair, delete accessory. But can not repair. There is not accessory in list.

maximkulkin, 2017-12-11:
There two cases here and you probably got one of them:
1. You paired it, then restarted, then unpaired while being "online" (connected to accessory). In that case controller (your iOS device) sends command to remove it's pairing record from accessory. The problem is that because currently library does not support reconfiguring mDNS (see [this issue](https://github.com/maximkulkin/esp-homekit/issues/3)), accessory continue to announce itself as not supporting pairing. You can restart accessory to fix that.
2. You removed pairing offline, so accessory still has your device pairing information and even if you restart it will not announce itself as supporting pairing. In that case I suggest erasing flash and reflashing firmware again (`make erase_flash flash`).

A handy command to master is mDNS query. On OS X it looks like this:
1. List accessories
```shell
> dns-sd -B _hap local
Browsing for _hap._tcp.local
DATE: ---Sun 10 Dec 2017---
23:40:28.581  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
23:40:28.884  Add        2   5 local.               _hap._tcp.           Sample LED
```
2. Show accessory info  - `dns-sd -L <accessory name> _hap local`
```shell
> dns-sd -L Sample\ LED _hap local
Lookup Sample LED._hap._tcp.local
DATE: ---Mon 11 Dec 2017---
 2:22:32.244  ...STARTING...
 2:22:33.962  Sample\032LED._hap._tcp.local. can be reached at Sample\032LED.local.:5556 (interface 5)
 md=Sample\ LED pv=1.0 id=69:AD:E1:AA:0E:9D c#=1 s#=1 ff=0 sf=1 ci=5
```
Notice that "sf=1".. That means that it allows pairing. If it is "sf=0" - it does not allow pairing.

HomeACcessoryKid, 2017-12-11:
you could also apply the /reset URL which should also wipe the storage and solve your issue (untested, based on code-review)

HomeACcessoryKid, 2017-12-11:
I think some redesign of that storage concept is in place and Max has already indicated he wants to move to spiffs. However, in case we stick to flash, the fact that there are 55 slots of 72 bytes means we have some space more than MAXPAIRINGS 16
So if a slot is disabled by killing the magic, it simply doesn't count.
Also, and to the point of this issue. Whenever removing a pairing (which must be done by an admin) count how many admins are left. If zero, reset the flash and start over automatically.
If run out of slots, consider storing all in RAM, reflash and store more compact.
I think that in a memory model where 512kB for the full code, spiffs might be too expensive so the flash model might be needed.
Insert your ideas below...

maximkulkin, 2017-12-13:
Ok, there was a bug with using dirty flash. I pushed a fix for that. Now it should properly use and reinitialize flash as needed. Admins are also taken into account: if no admin pairings left, it will switch back to pairing mode.

closed 2017-12-13 by maximkulkin