maximkulkin/esp-homekit-demo #6: dynamically generated characteristic ?

faronov, 2017-12-12:
Is it possible to generate characteristic
I try to explain
if I have multiple one wire temperature sensor like ds18b20
If I have one sensor I can write code,
if I want to add more, I need to recompile code for supporting next sensor
maybe it's possible to add some cycle that sees how many onewire sensors I have
and add characteristic for new one
This is just crazy idea :)

maximkulkin, 2017-12-12:
That is definitely possible to populate homekit structures in runtime. There is one limitation: to support declarative (static) definitions, I had to declare accessory's services and service's characteristics as a finite array of pointers. Currently, they can hold only 7 entries max. So, each accessory can have no more than 7 services and each service - no more than 7 characteristics. I know it is a limitation but good enough for now.

On the other hand, you can have (virtually) unlimited number of accessories, because you provide an array of homekit_accessory_t pointers to HomeKit server and you can allocate as many as you want. Just make sure that last item in array is a NULL pointer.

closed 2017-12-13 by maximkulkin

medmile, 2018-05-23:
Many thanks!