maximkulkin/esp-homekit-demo #7: Valid Values

medmile, 2017-12-13:
Haw can i set valid values for characteristic:
homekit_characteristic_t TARGET_HEATING_COOLING_STATE = {
    .type = HOMEKIT_CHARACTERISTIC_TARGET_HEATING_COOLING_STATE,
    .format = homekit_format_uint8,
    .permissions = homekit_permissions_paired_read
                 | homekit_permissions_paired_write
                 | homekit_permissions_notify,
    .min_value = (float[]) {0},
    .max_value = (float[]) {2},
    .min_step = (float[]) {1},
    0,
    1,
    2,
};

maximkulkin, 2017-12-13:
That is not supported yet, but I'm working on it and should finish pretty soon. Stay tuned.

maximkulkin, 2017-12-15:
Added support for valid-values and valid-values-range. Usage is like this:
```
homekit_characteristic_t TARGET_HEATING_COOLING_STATE = {
    .type = HOMEKIT_CHARACTERISTIC_TARGET_HEATING_COOLING_STATE,
    .format = homekit_format_uint8,
    .permissions = homekit_permissions_paired_read
    | homekit_permissions_paired_write
    | homekit_permissions_notify,
    .min_value = (float[]) {0},
    .max_value = (float[]) {2},
    .min_step = (float[]) {1},
    .valid_values = {
        .count = 3,
        .values = (uint8_t[]) { 0, 1, 2 },
    }
};
```

closed 2017-12-15 by maximkulkin