maximkulkin/esp-homekit-demo #8: Connection after wifi restart

medmile, 2017-12-13:
I test my device. And after my wifi router restart, ios can not connect to device. Can you help me?
This repeated:
{"characteristics":[{"aid":1,"iid":13,"value":22}]}
>>> client_send: Encrypting payload
>>> send_characteristic_event: Sending EVENT
>>> send_characteristic_event: Payload: {"characteristics":[{"aid":1,"iid":10,"value":28}]}
>>> send_characteristic_event: Sending EVENT: EVENT/1.0 200 OK
Content-Type: application/hap+json
Content-Length: 51

maximkulkin, 2017-12-16:
There seem to be problems in SDK a) handling loss of connectivity; b) restarting device after a crash.

Will keep looking into this.

maximkulkin, 2018-01-11:
Ok, after reviewing stack trace again it looks like it was out of memory problem. So, after loosing and restoring connectivity to accessory iOS tried to reconnect and thus re-request accessory information which by that time was too memory hungry. The memory usage was greatly optimized since then.

I will close this for now. If you feel there is still a problem, feel free to reopen it.

closed 2018-01-11 by maximkulkin