maximkulkin/esp-homekit-demo #9: Captive portal or web interface to set WiFi SSID and password

renssies, 2017-12-20:
Would it be possible (especially memory wise) to implement something like the Arduino WiFiManager that allows for setting the WiFi SSID and password using a simple web interface? 

I want to gift some small HomeKit sensors and devices to friends and family, but of course, you don't want the wifi to be hardcoded in those cases.

maximkulkin, 2017-12-20:
It is definitely on the radar but is kind of out of scope of HomeKit library. You could try merging HomeKit with [wificfg](https://github.com/SuperHouse/esp-open-rtos/tree/master/examples/wificfg)

renssies, 2018-01-03:
Thanks, I'm going to give that a try

maximkulkin, 2018-01-09:
I recently developed my own minimalistic captive portal library (because wificfg seemed like overcomplicated solution). You might want to give it a try: https://github.com/maximkulkin/esp-wifi-config

closed 2018-01-09 by maximkulkin

renssies, 2018-01-09:
Great, I've been experimenting with Wificfg for a few nights, but couldn't get it to work. 

Your esp-wifi-config however seems to cause a crash

renssies, 2018-01-09:
I'm using the following code (basically an edited version of the led example). The board is Wemos D1 Mini a 4MB (32Mb) ESP12E.
```C
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi_config.h"

const int led_gpio = 2;
bool led_on = false;

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void led_init() {
    gpio_enable(led_gpio, GPIO_OUTPUT);
    led_write(led_on);
}

void led_identify_task(void *_args) {
    for (int i=0; i<3; i++) {
        for (int j=0; j<2; j++) {
            led_write(true);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            led_write(false);
            vTaskDelay(100 / portTICK_PERIOD_MS);
        }

        vTaskDelay(250 / portTICK_PERIOD_MS);
    }

    led_write(led_on);

    vTaskDelete(NULL);
}

void led_identify(homekit_value_t _value) {
    printf("LED identify\n");
    xTaskCreate(led_identify_task, "LED identify", 128, NULL, 2, NULL);
}

homekit_value_t led_on_get() {
    return HOMEKIT_BOOL(led_on);
}

void led_on_set(homekit_value_t value) {
    if (value.format != homekit_format_bool) {
        printf("Invalid value format: %d\n", value.format);
        return;
    }

    led_on = value.bool_value;
    led_write(led_on);
}

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyLED"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
            HOMEKIT_CHARACTERISTIC(
                ON, false,
                .getter=led_on_get,
                .setter=led_on_set
            ),
            NULL
        }),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void on_wifi_ready() {
    printf("Connected to WiFi\n");
    led_init();
    homekit_server_init(&config);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    wifi_config_init("sampleled", NULL, on_wifi_ready);
}
```

Monitor gives me the following output
```
bpcll`pp_task_hdl : 3fff2508, prio:14, stack:512
pm_task_hdl : 3fff29f8, prio:1, stack:176
frc2_timer_task_hdl:0x3fff5e68, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jan  9 2018 23:22:49
phy ver: 273, pp ver: 8.3

rtos assert /Volumes/case-sensitive/esp-open-rtos/FreeRTOS/Source/queue.c 1434
abort() invoked at 0x402047b3.

0x402047b3: vAssertCalled at /Volumes/case-sensitive/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:233

Stack: SP=0x3fff48c0
0x3fff48c0: 3fff6168 00000000 00000000 402047b6

0x402047b6: vAssertCalled at ??:?

0x3fff48d0: 3fff6168 3fff4aac 3fff5fc4 40204f88

0x40204f88: xQueueSemaphoreTake at /Volumes/case-sensitive/esp-open-rtos/FreeRTOS/Source/queue.c:2374

0x3fff48e0: 3fff4aac 3fff4aac 3fff4910 3fff5fc4
0x3fff48f0: ffffffff 3fff4aac 3fff5fc4 402140d9

0x402140d9: _fflush_r at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdio/../../../../../newlib/libc/stdio/fflush.c:282

0x3fff4900: 40106230 3fff4aac 3fff5fc4 4023e5ec

0x40106230: __sfputs_r at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdio/../../../../../newlib/libc/stdio/nano-vfprintf.c:412

0x4023e5ec: json_flush at ??:?

0x3fff4910: 00000009 00000000 3fff1784 4020f6f8

0x4020f6f8: sysparam_get_data at /Volumes/case-sensitive/esp-open-rtos/core/sysparam.c:659

0x3fff4920: 3fff4aac 00000003 0000000a 40106280

0x40106280: __sfputs_r at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdio/../../../../../newlib/libc/stdio/nano-vfprintf.c:433

0x3fff4930: 3fff4aac 3fff4aac 3fff5fc4 3fff5fc4

Free Heap: 39116
_heap_start 0x3fff1a98 brk 0x3fff6234 supervisor sp 0x3ffffb00 sp-brk 39116 bytes
arena (total_size) 18332 fordblks (free_size) 0 uordblocks (used_size) 18332

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?

tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
```