maximkulkin/esp-homekit-demo #10: Error during paring

medmile, 2017-12-28:
After last update i have paring error each time:

dhcp client start...
ip:172.20.10.8,mask:255.255.255.240,gw:172.20.10.1
>>> HomeKit: Got IP, starting
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)

maximkulkin, 2017-12-28:
Pair setup is a memory expensive operation which requires ~20K of free RAM. How much free RAM do you have?

medmile, 2017-12-28:
I test temperature sensor sample and try to make erase_flash (nodemcu v3)

maximkulkin, 2017-12-28:
Please build it with HOMEKIT_DEBUG=1 to see your memory settings:
```
export HOMEKIT_DEBUG=1
make -C examples/temperature_sensor clean
make -C examples/temperature_sensor test
```

Also, make sure you update homekit component to latest version.

jmarchewka, 2018-01-02:
I was having the same problem (using HOMEKIT_DEBUG=1 showed a heap size of just over 20KB at the SRP generation failure point) and updating the esp-open-rtos library to the most recent version seems to have optimized memory usage somewhere (heap size showed almost 30K afterwards.) Everything seems to be stable with it now.

maximkulkin, 2018-01-02:
This is a tricky problem to address, cryptography does have high memory requirements and sensitive to fragmentation. Usually you can pair right after start, assuming your own tasks do not consume much memory. But there is a case when you remove last admin pairing during runtime and need to repair. By that time memory can be fragmented and/or used. Also, it makes “late initialization” approach less viable.

I’m playing with idea of writing a custom memory allocator for cryptography.

@HomeACcessoryKid I do not see much value in running memory reporting task unless you’re debugging cryptography internals since there is not much you can do to affect it. I debug cryptography by providing custom tracing implementations for XMALLOC/XFREE/...

HomeACcessoryKid, 2018-01-03:
After deleting the ultimate admin, you can overwrite the magic value and reset the app. Like this it always starts perfectly fresh. You really can split the code in two parts, before and after being paired. 
My code base has this covered so use it for details.

maximkulkin, 2018-01-10:
Closing this. Solution is to updated to latest esp-open-rtos which has more RAM available for allocation.

closed 2018-01-10 by maximkulkin