maximkulkin/esp-homekit-demo #11: Failed boot

faronov, 2017-12-29:
If I compile with make test, it works
but after restarted all demo wont boot
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

ets_main.c 
```

maccoylton, 2017-12-29:
do a flash erase and then try re-flash, in my experience this means the flash has't written correctly or been written to the wrong address

faronov, 2017-12-29:
I erase_flash and reflash
but this didn't help

faronov, 2017-12-29:
I try some examples from esp-open-rtos
like dht_sensor
it works perfectly
and also works after restart and(or) reset

faronov, 2017-12-29:
```
ip:192.168.11.23,mask:255.255.255.0,gw:192.168.11.1
>>> HomeKit: Got IP, starting
>>> HomeKit: Configuring mDNS
ip:192.168.11.23,mask:255.255.255.0,gw:192.168.11.1
>>> mdns_recv: pbuf too big
>>> mdns_recv: pbuf too big
>>> mdns_recv: pbuf too big
>>> mdns_recv: pbuf too big
```
what is mean pbuf too big ?

maximkulkin, 2017-12-30:
"pbuf too big" means that mDNS request message was too big to handle and was dropped. mDNS library I use tries to have smaller memory footprint, thus it puts constraints on input buffers. AFAIK should not cause too much problems.

jmarchewka, 2018-01-01:
@faronov I had this same problem using an ESP-01. The units I had have a 4MB flash and the examples default to a 32MB flash. If you edit the Makefile of the example and change the FLASH_SIZE variable to match your board, it should work. Good luck!

HomeACcessoryKid, 2018-01-02:
@jmarchewka, please note that B stands for BYTE and b for bits. ESP-01 uses 4 Mb and ESP-12 uses 4MB which is 32Mb

HacK

jmarchewka, 2018-01-02:
@HomeACcessoryKid so I was wrong before, in that my ESP-01 uses a EON_EN25Q80 for flash memory which, according to the [datasheet](http://pdf1.alldatasheet.com/datasheet-pdf/view/313132/EON/EN25Q80A-100HIP.html), is 8 Mb (1MB). The [-flash_size parameter for esptool.py](https://github.com/espressif/esptool#flash-size---flash_size--fs) is specified in bytes. If I set FLASH_SIZE to 1, the elf2image mode of esptool.py will fail (see below). So I am a bit confused about what exactly is happening (ie, why 4 or 8 works but 32 doesn't).

----------
CC /home/pi/homekit/esp-homekit-demo/examples/temperature_sensor/temperature_sensor.c
AR build/program.a
LD build/temperature_sensor.out
FW firmware/temperature_sensor.bin
usage: esptool elf2image [-h] [--output OUTPUT] [--version {1,2}]
                         [--flash_freq {40m,26m,20m,80m}]
                         [--flash_mode {qio,qout,dio,dout}]
                         [--flash_size {4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                         input
esptool elf2image: error: argument --flash_size/-fs: invalid choice: '1m' (choose from '4m', '2m', '8m', '16m', '32m', '16m-c1', '32m-c1', '32m-c2')
/home/pi/esp-open-rtos/common.mk:243: recipe for target 'firmware/temperature_sensor.bin' failed

HomeACcessoryKid, 2018-01-02:
esptool uses bits to specify flash size (suggested by m instead of M), so I think you should specify 
```
FLASH_SIZE = 8
```
I myself have just made an ZemiSmart RGBW bulb to work which also has 25Q80
I had to do erase_flash first before it started to behave
and I used
```
export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7A000
```
to define the location of the pairing info.

What is odd is that I have an ESP-01 from 2015 and it uses a 4Mb flash...

faronov, 2018-01-03:
I use esp-07 and looks like it also have 8m flash size

closed 2018-01-12 by faronov