maximkulkin/esp-homekit-demo #17: Shades support?

tompanhuhu, 2018-01-12:
Sorry if this is not an issue not used to github. 

Is it possible to control shades with this?
Making that homekit see's the device as a shade and is able to pass open/close/% data to the controller?

renssies, 2018-01-12:
You could build it yourself using an ESP and a stepper motor. I’ve built a roller shades version with these 3D printed parts: https://www.thingiverse.com/thing:2392856

You will also have to write your own example code though, mine still works over MQTT.

tompanhuhu, 2018-01-12:
Yes I have made an arudino board and can control the motor, it works fine but im looking for homekit integration and would like to have a standalone version and not be dependent on homebridge :) 

Very nice mount! Will check that out thanks!

HomeACcessoryKid, 2018-01-12:
Please download the homekit specification and check out sections 9.24 (Window Covering Service) and 8.47 (position), 8.52 (dynamic state) and 8.61 (%)
Then in the esp-homekit repo you can find the rest of the details in line with the esp-homekit-demo examples
Keep us posted?

renssies, 2018-01-12:
Well If you have any questions about shades in HomeKit I will be happy to help. I made a MQTT homekit integration for the shades as well

closed 2018-01-16 by maximkulkin