maximkulkin/esp-homekit-demo #18: can be this be compiled on ES32 ?

plkumar, 2018-01-12:
Can this be complied to work on ESP32? what changes needs to be done?

renssies, 2018-01-12:
No, this will not work directly on ESP32 because esp-open-sdk and esp-open-rtos don't work on ESP32 (yet). 

There are however people working on an ESP32 version of this library using the official ESP32 SDK. You can also find these on GitHub. The last time I checked they didn't work yet, they compile but most of them don't have the HomeKit pairing built yet.

jukorp, 2018-01-14:
I am very interested in esp32 and nrf52.
These are much quicker.
Esp idf. Can you build with this?

Resources should be directed to this new esp32 chip.

faronov, 2018-01-16:
https://github.com/aanon4/HomeKit nrf51 and nrf52 repo

maximkulkin, 2018-01-17:
So, currently the answer is NO. Closing this. You can proceed with further discussion in #23.

closed 2018-01-17 by maximkulkin