maximkulkin/esp-homekit-demo #19: Won't compile due to missing lip/dhcp.h

ForceFedDSM, 2018-01-13:
I have tried to compile the led example and I am getting an error that lwip/dhcp.h can't be found as required by user_main.c.  These files seem to be part of the esp-open-rtos and not yours so maybe this isn't where the issue should be raised.  I looked in user_main.c and it does have the include line in there.  I also checked the path and that file isn't there.  I cloned your repository using --recursive, did I do something wrong?  Sorry, I'm a little new to some of this stuff.

maximkulkin, 2018-01-13:
I guess you did not clone recursively esp-open-rtos repository. It also has submodules

ForceFedDSM, 2018-01-16:
Yes, you are correct.  Thank You.  I am now having an issue flashing to my hardware.  Make flash or test don't seem to be defined in the makefile. It gives me an error 'no rule to make target'

renssies, 2018-01-16:
@ForceFedDSM Have you set the SDKPATH to `esp-open-rtos` and added `esp-open-sdk/xtensa-lx106-elf/bin` to PATH?

If it does not work, could give us the full command you use (Including the directory terminal is currently in)

faronov, 2018-01-16:
no rule to make target mean that this file is not included
`include $(SDK_PATH)/common.mk `
probably you not set esp-open-rtos as $SDK_PATH

ForceFedDSM, 2018-01-16:
I have set my SDK_PATH to point to the esp-open-rtos. I will verify when I get home. I will run a printenv and post results. Maybe I had a typo but I checked and checked again. Also deleted and reloaded all components to try fresh. Still no go.

ForceFedDSM, 2018-01-17:
@renssies OK.  Confirmed that PATH and SDK_PATH are set correctly. Here are printenv results:

**PATH=/home/openhabian/esp-open-sdk/xtensa-lx106-elf/bin/:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games**

**SDK_PATH=/home/openhabian/esp-open-sdk/esp-open-rtos/**

My directory structure is /esp-open-sdk/esp-open-rtos/esp-homekit-demo

If I am in the /esp-open-sdk/esp-open-rtos/esp-homekit-demo/ directory and type "make -C examples/xxx all" I can compile the different examples. I get the successful output that says .bin file has been written to a directory.

Right after I get a successful compile I run "make flash" or "make test" within the /esp-open-sdk/esp-open-rtos/esp-homekit-demo/ directory.  This is where I get the error.  

Do I need to type additional arguments with "make flash"?

ForceFedDSM, 2018-01-17:
I can't believe I was missing that.  Yep, I needed to type all the arguments.  Just flashed my chip.  Now to test.  Thanks all for the help.

maximkulkin, 2018-01-17:
You can actually go to particular example’s directory and just call “make flash” there. Make’s “-C <dir>” option just instructs make to change directory before normal execution.

closed 2018-01-17 by maximkulkin