maximkulkin/esp-homekit-demo #21: multiple sonoff

faronov, 2018-01-14:
First, i have problem with debugging sonoff example
it won't boot in low power mode (from firmware update USB stick)
wifi_config seems nice (and working) feature
but If I try to use multiple this device
I have worked only one, the previous pairing seem to fail after new pairing 
and if I can't debug this, I thought maybe problem with generating device UUID or certificate?
or maybe this problem with mDNS names?

faronov, 2018-01-14:
I still have too slow pairing issue (about 1-3 minutes !!!) this is very slow and when I use the last version of HomeACcessoryKid repo, it works MUCH faster, like 5-10 seconds !!!!
also, I have a false pairing sometimes

renssies, 2018-01-14:
Not booting from USB could be related to the max USB power available. Try a USB 3.0 port directly on your computer. If you're using a PC, using a USB hub might help. Some motherboards don't supply enough power over USB.

As for the multiple Sonoff devices, make sure the accessory information name is unique. I've opened a PR for that earlier today.

faronov, 2018-01-15:
Don't know that information name must be unique
about booting, look strange, because it works before 
One more question regarding SonOff
seems led for relay state disabled

maximkulkin, 2018-01-16:
Ok, the issue was solved by implementing support custom accessory names, is that correct? OK to close?

faronov, 2018-01-16:
yes issue resolved

closed 2018-01-16 by faronov