maximkulkin/esp-homekit-demo #22: Initial State not reported after power cycle

Arduingo, 2018-01-14:
I don't know if this is an issue, I hope some one may help my clarify this behavior:

Using de Led or Sonoff switch examples:

1) Power On and Pair, and turn it ON, the status is reflected on the home app
2) Power off the ESP by disconnecting from the power supply
3) Power On again, both examples will fall back to off status, but in the home app the status stays as ON
4) Close the home app, open again, the status is refreshed after few seconds, the log shows a pair and characteristics updates

I tested with commercial HomekIt solutions and this is not the case for example HUE lamps that after a power lost fall back to ON status, the status is refreshed in the home App with no problem in a very short time.

After studied the sonoff example (with my limited knowledge on this topic) I tried by including a homekit_characteristic_notify during gpio_init, but no luck, interesting is that the same homekit_characteristic_notify in the interrupt routine to detect button activation works perfect and the status is updated.

Is this an expected behavior?

renssies, 2018-01-14:
I can't say if it's expected behavior, but it's true that it's currently not built in. Not sure what we should do here. I'm not a complete expert on the HomeKit specification. 

The reason adding `homekit_characteristic_notify` in `void gpio_init()` doesn't work is because the HomeKit server hasn't started yet. If you look at the code you will see that the HomeKit server is started when the wifi is ready (in `void on_wifi_ready()`). If you call it there I'm pretty sure that it will work :)

maximkulkin, 2018-01-14:
The specification does not describe behavior in this case and I guess each
accessory picks whatever works best. E.g. my LIFX light bulb always turns
on after power cycle.

The framework itself does not handle this partly because it is very easy to
integrate any strategy without changing the framework itself: you just need
to save state when it is changed and restore it on init. But keep in mind
that this will wear off flash memory (that’s why framework does not do
this).

I can post you example when I’ll get home, it’s just a couple of lines.

renssies, 2018-01-14:
I just researched some things because I was curious

> I tested with commercial HomekIt solutions and this is not the case for example HUE lamps that after a power lost fall back to ON status, the status is refreshed in the home App with no problem in a very short time.

- Indeed Hue does notify HomeKit when a light turns on, but the bridge also has more power and way better flash (SSD grade flash actually).
- My Koogeek outlet does not notify HomeKit of its state when power returns, same for My LIFX bulb. 

So I'm sure this is not a rule or requirement that is part of the (commercial) MFI specification and just something Philips has implemented.

> But keep in mind that this will wear off flash memory (that’s why framework does not do
this).

I once had a discussion about this in real life, apparently, the flash on the cheap Wemos D1 mini is rated at a minimum of 100,000 write cycles. Same as the official Arduino Uno. Of course, there are knockoff ESP modules, but the lowest failure rate people have seen is ~10,000 write cycles. 

This, of course, doesn't mean we should implement state restoration in the framework.

Arduingo, 2018-01-14:
Thanks to both for your answers, @maximkulkin , I appreciate if you could show me an example, I agree this should not be handled by the framework and each vendor may define their own strategy, what I´m looking is a way to reflect the correct status, I´m trying to mimic the LIXF and HUE behaviour  but as @renssies states I very likely using the wrong place to notify the status to homekit.

maximkulkin, 2018-01-14:
Exactly. As of notifications: indeed, by the time gpio_init is run, there
are no clients connected. Usually when clients connect they request current
state on their own, you do not need to do anything. Just make sure that you
put correct value into characteristic.

Arduingo, 2018-01-15:
I think understand now what is happening, in the case of HUE, the lamp is the one who lost power but the bridge is the one to were the client is connected so connection is never lost and the notification works, in my case the server lost connection like the koogeek outlet, so until the client home app do not connect again or request an update (don´t know if this happens periodically) is not possible to start getting notifications or even get the new status in the case is stored correctly into the characteristic. i`m correct?

maximkulkin, 2018-01-15:
Afterthought: the main problem with reading state I see is when initial
attempt fails (e.g. device is powered off, hung, etc), the next attempt is
done after only some period. Not sure if there is a way to signal iOS
controller to redo connection faster.

maximkulkin, 2018-01-16:
So, the code is like this:
```c
#include <sysparam.h>
#define LIGHT_ON_PARAM "light_on"

void light_on_change(homekit_characteristic_t *on, homekit_value_t value, void *context) {
    sysparam_set_bool(LIGHT_ON_PARAM, value.bool_value);
}

homekit_characteristic_t light_on = HOMEKIT_CHARACTERISTIC_(
    ON, false,
    .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(light_on_change)
);

void light_inti() {
    sysparam_get_bool(LIGHT_ON_PARAM, &light_on.value.bool_value);
    // rest of init
}
```

Arduingo, 2018-01-16:
Thanks for the example!

closed 2018-01-16 by Arduingo

HomeACcessoryKid, 2018-01-16:
@maximkulkin 

> Afterthought: the main problem with reading state I see is when initial
attempt fails (e.g. device is powered off, hung, etc), the next attempt is
done after only some period. Not sure if there is a way to signal iOS
controller to redo connection faster.

There are two ways to force an update:
- use the control center, select home. To refresh, go one up, then select home again
- use eve (el-gato) which has a drag-down-to-update concept

But the home app is as passive as they get