maximkulkin/esp-homekit-demo #23: Port to ESP_IDF

zafrirron, 2018-01-15:
ESP_IDF is considered to be the standard for ESP32 SDK toolchain source and API's
Would you consider porting this great work to ESP_IDF?

thanks

maximkulkin, 2018-01-15:
I'm already working on a port. There are some problems along the way. E.g. ESP_IDF is based on 2 year old LWIP library while esp-open-rtos has a very recent version and esp-homekit uses features from recent releases. I really do not want to completely fork esp8266 and esp32 implementations, so right now I will wait for ESP_IDF to catch up (one of core devs said that they have their own customizations for LWIP but they work on porting them to new version).

zafrirron, 2018-01-16:
Thanks!
sounds like a good plan! 
I will also open issue on ESP_IDF to try to push...

affonsoloyola, 2018-02-11:
I’m trying really hard to make this compile with IDF. I’m struggling to adapt the components structure to the default structure of IDF. About lwip. I think you can override the IDF lwip. Also I think the crutpto component included with IDF could also replace wolf crypt. What do you think? Do you have something partially working on ESP32?

jukorp, 2018-03-15:
Good news. ESP-IDF lwip update coming. 1.5.0 -> 2.0.3
https://github.com/espressif/esp-idf/issues/1512

jukorp, 2018-03-15:
Are there other problems than the lwip version? wolfssl?

maximkulkin, 2018-05-09:
Finally this is finished to a usable degree. Check it out

closed 2018-05-09 by maximkulkin