maximkulkin/esp-homekit-demo #25: Other devices and license questions

RavenSystem, 2018-01-26:
Hello @maximkulkin, 

I'm a HomeKit fan, and I'm following all your work. It's amazing!

I'm using your actual work to code final versions of firmwares that can be distributed in binary format to let users to install them without a dev environment. For now, I have a Sonoff Basic with some extra functions: a reset option pushing quickly the button, and a option to connect a wall switch to GPIO14 and ground. And actually, I'm working with a Sonoff TH to convert it to a HomeKit thermostat. Please, can you tell me if my work is right with license terms.

Many thanks.

maximkulkin, 2018-01-26:
Can you write me an email (it's on my Github account).

maximkulkin, 2018-01-27:
Well, the license is MIT which is kind of permissive although it requires you to include original LICENSE with your distribution.

RavenSystem, 2018-01-27:
Many thanks, 

I have updated LICENSE. You can check it here: https://github.com/RavenSystem/esp-homekit-devices/blob/master/LICENSE

closed 2018-01-29 by maximkulkin