maximkulkin/esp-homekit-demo #27: Is this project able to be built/run on windows?

wmwood, 2018-01-29:
I'm pretty new to this I was hoping to find info on how to get the required tools installed and going on windows.

Any tutorial?

renssies, 2018-01-29:
You should be able to run it on Windows, although it will be harder. Building the project requires a GNU/POSIX based system according to esp-open-sdk. But apparently, you can use Cygwin. You could try to look for tutorials to install `make` and build `esp-open-sdk` on windows.

However, you might be better off by running Linux (I recommend Ubuntu) in a virtual machine with 2GB ram.

renssies, 2018-01-29:
Correction: turns out there are a lot of issues with building on Windows. Even Microsoft recommends, installing Linux. In Windows 10 this can be done through the Microsoft store: https://docs.microsoft.com/en-us/windows/wsl/install-win10

Arduingo, 2018-01-30:
My recommendation is use VirtualBox as a Virtual Machine and install Ubuntu Server on it, the install all the tools and SDK on the Vistual machine, it takes some time for install but will save you lot of pain in the future. Here a couple of excellent tutorial on the whole process: https://youtu.be/kYEzEFH5LiM, https://youtu.be/T-oSjMCmNYk, For the second part remember you first read the Open-RTM documentation for the Open-SDK requirements.

ysdndev, 2018-02-06:
I too am pretty new to this. However, not sure where to ask this, but working on Mac and having trouble making example/led after having followed setup instructions as well as I've been able to decipher them.

`make -C examples/led all`

seems to have worked fine, as did 

`export ESPPORT=/dev/tty.wchusbserial1420`

but when I run

`make -C examples/led test`

the flash process seems to get to 100% and then I get the following 'Fatal exception (28)...' repeating infinitely until I cancel out and receive the following

`Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
^CTraceback (most recent call last):
  File "/Volumes/case-sensitive/esp-open-sdk/esp-open-rtos///utils/filteroutput.py", line 105, in <module>
    main()
  File "/Volumes/case-sensitive/esp-open-sdk/esp-open-rtos///utils/filteroutput.py", line 96, in main
    addr2line = subprocess.check_output(["xtensa-lx106-elf-addr2line","-pfia","-e","%s" % args.elf, addr], cwd=".").strip()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 566, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1231, in _execute_child
    self.pid = os.fork()
KeyboardInterrupt
make: *** [test] Error 1
`

I'm not sure if it is my setup or something I have missed. Any help would be greatly appreciated.

Again, my apologies if this isn't the proper place for this sort of question.

maximkulkin, 2018-02-06:
@ysdndev What kind of device do you use? Did you properly set FLASH_SIZE and thus HOMEKIT_SPI_FLASH_BASE_ADDR?

maximkulkin, 2018-02-06:
Ok, I've updated README with additional instructions on how to configure project according to size of ESP flash.

ysdndev, 2018-02-06:
I am using a LoLin dev board which indicates NodeMCU v3. According to most searches, it is based on the ESP8266 12E. Not sure what other info you might need. Thanks for the response.

maximkulkin, 2018-02-06:
@ysdndev Try building firmware as if it has only 1MB of flash (following updated README instructions). Also, when you provide error output, please include more output from MCU itself, not from serial communication tool (filteroutput.py in this case)

ysdndev, 2018-02-06:
I managed to flash the sonoff_basic example which works great, though I still get the same issues trying to flash the led example even as 1Mb. When I try filteroutput.py I get 'command not found'.

ysdndev, 2018-02-06:
I found last comment by @carmanuelz in maximkulkin/esp-homekit#5 which allowed me to properly flash the led example. Thanks for the help!

closed 2018-02-06 by maximkulkin