maximkulkin/esp-homekit-demo #28: stepper delay

MotoManiacMM, 2018-01-31:
Hi, I'm trying to make door lock with stepper motor. I implemented code for stepper instead of led in your user_main.c, but I need delay of 2ms between the steps and vTaskDelay is 10ms. I tried to change configTICK_RATE_HZ from 100 to 1000 in FreeRTOSConfig.h, but with no effect. Can you please help me with this? Can I change vTaskDelay to 1ms? or can I use some other way to make 2ms pauses between steps? thank you!

maximkulkin, 2018-01-31:
This is kind of not the right place to ask this (you should go to [esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos) instead), but I can see that DHT driver there uses 2 microsecond delays using [sdk_os_delay_us()](https://github.com/SuperHouse/esp-open-rtos/blob/2ab9beb946521a918e8cb69d802e732afa7cb9d8/extras/dht/dht.c#L67). That does not play well with FreeRTOS tasks, but at least you can get what you want.

MotoManiacMM, 2018-02-01:
THANK YOU!!! It solved my issue!!!

closed 2018-02-01 by maximkulkin