maximkulkin/esp-homekit-demo #30: Button debounce behavior in button examples

jreed509, 2018-02-10:
In the current button example, debouncing does not behave quite properly. 

After the first edge, the interrupt handler ignores further edges for the debounce period - this makes sense. However, immediately after the first edge is detected the button GPIO state is read to determine if it is pressed or depressed. This can result in reading the state of a "bounce" rather than the stable button state and cause undefined button behavior. 

I solved this in my code by adding `sdk_os_delay_us(1000*button->debounce_time);` before the if statement to determine state. 

This seems to give much more stable button response, although I realize adding hard delays in the interrupt handler isn't ideal.

maximkulkin, 2018-02-12:
Apart from sleeping in interrupt handle, not sure how waiting more helps detecting what was the first edge.

jreed509, 2018-02-13:
So adding the delay doesn't help detecting which was the first edge (this works great now), but it does help when determining whether the button is pressed or released. 

[image]

Currently immediately after seeing the first edge, the interrupt handle checks the GPIO state to determine if that edge corresponds to a button "press" or button "release". This is shown in the lower signal in the diagram above. Because we check immediately, it is possible that the signal has not stabilized and the incorrect state will be read. 

Maybe the easiest way to handle this is to have a flag that stores the button's pressed state - i.e. a button can only be released if it was previously pressed. Something like this:
```
typedef struct _button {
    ...
    bool is_pressed;
    ...
} button_t;

void button_intr_callback(uint8_t gpio) {
    button_t *button = button_find_by_gpio(gpio);
    if (!button)
        return;

    uint32_t now = xTaskGetTickCountFromISR();
    if ((now - button->last_event_time)*portTICK_PERIOD_MS < button->debounce_time) {
        // debounce time, ignore events
        return;
    }
    button->last_event_time = now;
    
    if (!button->is_pressed) {
        // Record when the button is pressed down.
        button->last_press_time = now;
        button->is_pressed = true;
    } else {
        // The button is released. Handle the use cases.
        if ((now - button->last_press_time) * portTICK_PERIOD_MS > button->long_press_time) {
            button->callback(button->gpio_num, button_event_long_press);
        } else {
            button->callback(button->gpio_num, button_event_single_press);
        }
        
        // Unset button state
        button->is_pressed == false;
    }
}
```

affonsoloyola, 2018-02-20:
I’m experimenting with this example and I’m not sure if it’s working as supposed. I see the internal pull-up being set on the code. So, a change of level to low would be simply achievable by having a momentary switch to ground to create an event. But this does not work with my circuit. If I connect a switch to ground the behavior of the button is inverted. If a I keep the button pressed furring startup and momentarily released it it create the single press event. If I connect the pin to a resistor connected to ground and also to a switch connected to vcc the button example works. I’m not sure what’s going on but I suspect the initial state of the button is being considered pressed ... I’ll do some more test ...

maximkulkin, 2018-03-05:
@affonsoloyola Any updates on this?

Just in case, when I wrote it and tested on buttons that I had, it worked perfectly. I guess, it depends on the type of buttons you use. My main intent was not to build a universal button library, but to show a) a good design for a button library (in my opinion of course, instead of cluttering main code with interrupt handling details); b) how you can implement read-only user triggered home kit characteristics (for which programmable button is a perfect case).

closed 2018-05-09 by maximkulkin