maximkulkin/esp-homekit-demo #32: Can't flash ESP8266

bobmosh, 2018-02-11:
Hey,

I'm working with a WeMos Lolin32 with an OLED display.
I've managed to get the LED example to run be made properly, and now I'm at the step of trying to flash the code to the ESP chip.

The command `make -C examples/led test`brings out the following output:

```
Ferdinands-MacBook-Pro:esp-homekit-demo Ferdi$ make -C examples/led test
esptool.py -p /dev/cu.SLAB_USBtoUART --baud 115200 write_flash -fs 8m -fm qio -ff 40m \
		0x0 /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/led.bin 
esptool.py v1.2
Connecting...

A fatal error occurred: Failed to connect to ESP8266
make: *** [flash] Error 2
```

The funny thing at this point is, that the chip is reset multiple times (3 to be precise) at the flashing process, so I can confirm that the connection works. Apparently the issue lies within the flashing. What am I doing wrong? Am I missing something here?

If you need any infos from me just ask, I don't know what could factor into this, so I don't know which exact informations to provide.

Arduingo, 2018-02-12:
if you run 

esptool.py flash_id 

what do you get?

maximkulkin, 2018-02-12:
You need to put chip into flash mode by connecting gpio0 to ground when you power it on. NodeMCU boards do that automatically but apparently your board doesn’t. There are two features those dev boards support: auto reset and auto flash. Looks like yours still support auto reset, but not auto flash.

maximkulkin, 2018-02-12:
Ok, i checked lolin32 schematic, it does support both auto features. Before digging into details, can you try updating esptool to latest version (2.x) and see if that helps.

maximkulkin, 2018-02-12:
One thing that I noticed: you use /dev/cu.SLAB_USBtoUART while I always use /dev/tty.SLAB_USBtoUART.

bobmosh, 2018-02-12:
@Arduingo, The Output of `esptool.py flash_id` is this:
```
Ferdinands-MacBook-Pro:esp-homekit-demo Ferdi$ esptool.py flash_id
esptool.py v1.2
Traceback (most recent call last):
  File "/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 1270, in <module>
    main()
  File "/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 1179, in main
    esp = ESPROM(args.port, initial_baud)
  File "/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 71, in __init__
    self._port = serial.serial_for_url(port)
  File "/Library/Python/2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
```
Looks like he's trying on the wrong port to me, since my board is on /dev/cu.SLAB_USBtoUART.

The update to 2.3 did the trick! Thanks a lot! Seems like I have 2 different versions installed on my machine and the PATH always pointed to the old version which created the issue. Thanks for the help!

closed 2018-02-12 by bobmosh