maximkulkin/esp-homekit-demo #33: Issue with flashing

bobmosh, 2018-02-12:
Hey,

sorry to bother again, but I'm having trouble to run the monitor command on all of the demos.
Again I'm using a WeMos Lolin32 and right now, I can properly make the code and (looks like it) correctly flash the code to the ESP, but the functionality isn't there. I was using the wifi_config example to create an AP from the ESP, but it won't happen. I've tried a bunch of things now, compared the commands with the one from the Arduino IDE wich are certain to work (even wit the wifi.h imported), and googled my way around the internet but can't exactly figure out whats going on here.

I'm using the following commands:
```
Ferdinands-MacBook-Pro:esp-homekit-demo Ferdi$ make -C examples/wifi_config all
make: Nothing to be done for `all'.

Ferdinands-MacBook-Pro:esp-homekit-demo Ferdi$ make -C examples/wifi_config flash
esptool.py -p /dev/cu.SLAB_USBtoUART --baud 115200 write_flash -fs 8m -fm qio -ff 40m \
		0x0 /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/wifi_config.bin 
WARNING: Flash size arguments in megabits like '8m' are deprecated.
Please use the equivalent size '1MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.3-dev
Connecting........__
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 126.7 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 2324.3 kbit/s)...
Hash of data verified.
Compressed 457968 bytes to 322494...
Wrote 457968 bytes (322494 compressed) at 0x00002000 in 28.5 seconds (effective 128.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Ferdinands-MacBook-Pro:esp-homekit-demo Ferdi$ make -C examples/wifi_config monitor
/Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos//utils/filteroutput.py --port /dev/cu.SLAB_USBtoUART --baud 115200 --elf ./build/wifi_config.out
Opening /dev/cu.SLAB_USBtoUART at 115200bps...
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
^CTraceback (most recent call last):
  File "/Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos//utils/filteroutput.py", line 105, in <module>
    main()
  File "/Users/Ferdi/Desktop/RISE/Code/ESP/ESP_examples/esp-open-rtos//utils/filteroutput.py", line 87, in main
    line = port.readline()
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
```

Any ideas what I am doing wrong here?

bobmosh, 2018-02-12:
Seems to be an issue with the esp-open-rtos, I can't even get simple blink examples to work. I'll try to get in touch with the people there and post my solution here afterwards!

renssies, 2018-02-12:
The wemos Lolin32 uses an esperrif micro controller that is part of the ESP32 platform.

This code is for the ESP8266 platform and is not compatible with ESP32.

bobmosh, 2018-02-12:
OMG really, am I this stupid? 😮

Thanks for the info @renssies! I'll get my hands on an ESP8266 and try again.

Sorry for the very much unneeded issue...

closed 2018-02-12 by bobmosh