maximkulkin/esp-homekit-demo #34: Documentation and understanding

bobmosh, 2018-02-16:
Hey,

I've now gotten my NodeMCU and tested the code on it. It works like a charm so far! Thanks so much for creating this library and the examples.

However, now that I want to get into developing some functionalities myself, I'm having some trouble getting the right information. Since the code is not documented at all, I can only assume what's happening in the different functions, etc.

I'm currently going through some of the examples and trying to create inline documentation. I'm also willing to publish them when they're done, but for now I need to get things straight in my head first!

Can anyone help me what these two functions then from the led example do?

```
// No idea what this does...
void led_identify_task(void *_args) {
    for (int i=0; i<3; i++) {
        for (int j=0; j<2; j++) {
            led_write(true);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            led_write(false);
            vTaskDelay(100 / portTICK_PERIOD_MS);
        }

        vTaskDelay(250 / portTICK_PERIOD_MS);
    }

    led_write(led_on);

    vTaskDelete(NULL);
}

// No idea what this does either...
void led_identify(homekit_value_t _value) {
    printf("LED identify\n");
    xTaskCreate(led_identify_task, "LED identify", 128, NULL, 2, NULL);
}
```

renssies, 2018-02-16:
It would be cool if you could make a PR with the documentation. It would help others :)

Let me explain the methods:

**led_identify**
```C
void led_identify(homekit_value_t _value) {
    printf("LED identify\n");
    xTaskCreate(led_identify_task, "LED identify", 128, NULL, 2, NULL);
}
```

This method starts the `led_identify_task` that runs until ` vTaskDelete(NULL);` is called. This method is required for a HomeKit accessory. If you look at the code on line 82: https://github.com/maximkulkin/esp-homekit-demo/blob/90693e10d94a1f19a3bd1737512e9cae91fc3428/examples/led/led.c#L82
You will see that it's part of the "Accessory Information" service in the configuration.

`identify` is called to let the user easily identify the accessory they are adding. This is performed automatically one time after the device has been paired but before it's actually added. There is also an `identify` button on that view so the user can call this more than once. 

**led_identify_task**
```C
void led_identify_task(void *_args) {
    for (int i=0; i<3; i++) {
        for (int j=0; j<2; j++) {
            led_write(true);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            led_write(false);
            vTaskDelay(100 / portTICK_PERIOD_MS);
        }

        vTaskDelay(250 / portTICK_PERIOD_MS);
    }

    led_write(led_on);

    vTaskDelete(NULL);
}
```
This task blinks the LED 2 times in 3 sequences. The delay between the blinks is 100ms, the delay between the sequences is 250ms.

bobmosh, 2018-02-16:
Allright, I'll definitely work on the documentation and create a PR afterwards, however I need to get into it first, before I can do that. :)

Thanks for the immediate feedback! So what you're describing there means that this is a visual feedback for the user? This has nothing to do with actual HomeKit functionality, like determining what type of accessory this is?

renssies, 2018-02-16:
Yes, this is only visual feedback to the user. See `5.7.5 Identify Routine` in the non-commercial HAP Specification: https://developer.apple.com/homekit/specification/

To give some examples:
- Philips Hue Bulbs blink twice
- The Philips hue bridge blinks it's LED ring (as I recall correctly)
- The Nanoleaf panels blink 3 times
- A LIFX bulb blinks 3 times
- A Koogeek socket blinks its built-in LED 3 times
- The onma camera blinks an LED on the front

closed 2018-03-05 by maximkulkin