maximkulkin/esp-homekit-demo #36: two buttons motor support

ageorgios, 2018-02-21:
I am trying to implement a two touch switch device that controls a motor. I am only starting though.

I am trying to figure out how to implement the interlocking between the two buttons.
To clarify, AT NO TIME should both switches be ON at the same time, because this may damage the motor. So it is crucial that there will be no possibility such a thing to happen.

I am wondering if such code works so as to exclude this kind of situation
```
void button1_task() {
    printf("UP start\n");
    gpio_write(down_gpio, 0);
    gpio_write(up_gpio, 1);
    vTaskDelay(4000 / portTICK_PERIOD_MS);
    gpio_write(up_gpio, 0);
    printf("UP stop\n");
    vTaskDelete(NULL);
}
void button1_callback(uint8_t gpio_num, button_event_t event) {
    printf("Button UP Pressed\n");
    if (down_task_handle != NULL) {
        printf("It is during moving DOWN, so stopping.\n");
        vTaskDelete(down_task_handle);
        gpio_write(down_gpio, 0);
    }
    else if (up_task_handle != NULL) {
        printf("It is during moving UP, so stopping\n");
        vTaskDelete(up_task_handle);
        gpio_write(up_gpio, 0);
    }
    else if (up_task_handle == NULL && down_task_handle == NULL) {
        xTaskCreate(button1_task, "UP Task", 128, NULL, 2, up_task_handle);
    }
}

void button2_task() {
    printf("DOWN start\n");
    gpio_write(up_gpio, 0);
    gpio_write(down_gpio, 1);
    vTaskDelay(4000 / portTICK_PERIOD_MS);
    gpio_write(down_gpio, 0);
    printf("DOWN stop\n");
    vTaskDelete(NULL);
}
void button2_callback(uint8_t gpio_num, button_event_t event) {
    printf("Button DOWN Pressed \n");
    if (up_task_handle != NULL) {
        printf("It is during moving UP, so stopping.\n");
        vTaskDelete(up_task_handle);
        gpio_write(up_gpio, 0);
    }
    else if (down_task_handle != NULL) {
        printf("It is during moving DOWN, so stopping\n");
        vTaskDelete(down_task_handle);
        gpio_write(down_gpio, 0);
    }
    else if (up_task_handle == NULL && down_task_handle == NULL) {
        printf("Not moving so going DOWN\n");
        xTaskCreate(button2_task, "DOWN Task", 128, NULL, 2, down_task_handle);
    }
}
```

marcoskirsch, 2018-02-22:
> To clarify, AT NO TIME should both switches be ON at the same time, because this may damage the motor. So it is crucial that there will be no possibility such a thing to happen.

I don't think software is the way to ensure this. Most relay you'd buy to control your motor already have 3 terminals - common, and two that alternate between on and off. Does the Sonos not provide the 3 terminals? Maybe look for something else.

ageorgios, 2018-02-23:
No it doesn't, both switches can be on.

maximkulkin, 2018-02-23:
@ageorgios Your question is about how to use multitasking synchronization properly, right? In your case it would be sufficient to use binary semaphore:

```c
#include <semphr.h>

static SemaphoreHandle_t mutex = NULL;

void user_init() {
    mutex = xSemaphoreCreateBinary();
    xSemaphoreGive(mutex);
    // ...
}

void button_callback(uint8_t gpio_num, button_event_t event) {
    if (xSemaphoreTake(mutex, 0) == pdTrue) {
        // Got mutex, proceed with processing
        // ...
        xSemaphoreGive(mutex);
    }
}
```

closed 2018-02-23 by ageorgios