maximkulkin/esp-homekit-demo #37: memory: stack overflow

ageorgios, 2018-02-21:
I am creating a TASK with xTaskCreate and I am getting stack overflow errors.
Is the memory so small that it could happen?

renssies, 2018-02-21:
What is your full code? You should be able to create a few tasks, but only after pairing is done

ageorgios, 2018-02-21:
xTaskCreate(buttonUP_task, "UP Task", 64, (void *) duration, 2, &up_task_handle);
xTaskDelete(NULL)

when calling it 5-6 times it needs a reset

maximkulkin, 2018-02-21:
What is the task code? "64" actually will allocate 64 32bit words which is 256 bytes. To prevent stack overflows you need to increase task stack size, not decrease it (e.g. change 64 to 256).

ageorgios, 2018-02-23:
When pressing button -> stack overflow...

```
void buttonDOWN_task(void * duration) {
    uint32_t dur = (uint32_t)duration;
    printf("DOWN start: %i\n",dur);
    gpio_write(up_gpio, 0);
    gpio_write(down_gpio, 1);
    vTaskDelay(dur / portTICK_PERIOD_MS);
    gpio_write(down_gpio, 0);
    printf("DOWN stop\n");
    down_task_handle = NULL;
    vTaskDelete(NULL);
}

void buttonDOWN_handle(uint32_t duration) {
    printf("Button DOWN Pressed \n");
    if (up_task_handle != NULL) {
        printf("It is during moving UP, so going DOWN.\n");
        vTaskDelete(up_task_handle);
        up_task_handle = NULL;
        gpio_write(up_gpio, 0);
        xTaskCreate(buttonDOWN_task, "DOWN Task", 256, (void *) duration, 2, &down_task_handle);
    }
    else if (down_task_handle != NULL) {
        printf("It is during moving DOWN, so stopping\n");
        vTaskDelete(down_task_handle);
        down_task_handle = NULL;
        gpio_write(down_gpio, 0);
    }
    else if (up_task_handle == NULL && down_task_handle == NULL) {
        printf("Not moving so going DOWN\n");
        xTaskCreate(buttonDOWN_task, "DOWN Task", 256, (void *) duration, 2, &down_task_handle);
    }
}

void button_callback(uint8_t gpio_num, button_event_t event) {
    if (xSemaphoreTake(mutex, 0) == pdTRUE) {
        if (gpio_num == button1_gpio) buttonUP_handle(4000);
        else if (gpio_num == button2_gpio) buttonDOWN_handle(4000);        
        xSemaphoreGive(mutex);
    }
    else {
        printf("race condition, doing nothing");
    }
}
```

maximkulkin, 2018-02-23:
Try localizing problem by disabling different parts of your code (e.g. disable mutex and see if task creation works, disable task creation and see if mutex works).

You could actually save a lot of friction if you would just create an always running motor control task and communicate with it via sending messages through a queue. So, your messages could be "set state to open" or "set state to closed" and it will handle the rest. Example:
```c
#include <queue.h>

static QueueHandle_t door_queue = NULL;

typedef enum {
    door_command_open,
    door_command_close
} door_command;

typedef  enum {
    door_state_open,
    door_state_closed,
    door_state_opening,
    door_state_closing,
} door_state;

void door_task(void *context) {
    door_state state = door_state_closed;
    while(1) {
        if (xQueueReceive(door_queue, &cmd, 0)) {
            switch(cmd) {
                case door_command_open: {
                    switch(state) {
                        case door_state_open:
                        case door_state_opening:
                            break;
                        case door_state_closing:
                            gpio_write(down_gpio, 0);
                        case door_state_closed:
                            gpio_write(up_gpio, 1);
                            state = door_state_opening;
                            break;
                    }
                    break;
                }
                case door_command_close: {
                    switch(state) {
                        case door_state_opening:
                            gpio_write(up_gpio, 0);
                        case door_state_open:
                            gpio_write(down_gpio, 1);
                            state = door_state_closing;
                            break;
                        case door_state_closed:
                        case door_state_closing:
                            break;
                    }
                    break;
                }
            }
        }
    }
}

void user_init() {
    door_queue = xQueueCreate(20, sizeof(door_command));
}

void send_door_command(door_command cmd) {
    xQueueSendToBack(door_queue, &cmd, 10);
}

void button_callback(uint8_t gpio_num, button_event_t event) {
    if (gpio_num == button1_gpio) send_door_command(door_open);
    else if (gpio_num == button2_gpio) send_door_command(door_close);
}
```

The code above is reactive so you can change door direction halfway through. Depending on your setup you need to implement either handling endstop signals or tracking how much time you spent opening or closing and in case of reverse you take that time into account.

ageorgios, 2018-02-24:
Great ideas you have there! Thank you, I have to sleep on it and start again with the Queue use you propose!

ageorgios, 2018-02-28:
in case someone wants a motor https://github.com/ageorgios/esp-homekit-motor

closed 2018-02-28 by ageorgios