maximkulkin/esp-homekit-demo #38: LED Example: On/Off Inverted in Homekit Display

brianhanifin, 2018-02-22:
When I ran the LED example Homekit showed that the Light was off when the LED was on (and vice versa). I got the expected result by modifying the following line:

https://github.com/maximkulkin/esp-homekit-demo/blob/90693e10d94a1f19a3bd1737512e9cae91fc3428/examples/led/led.c#L29

`gpio_write(led_gpio, on ? 1 : 0);`

The way you have it currently the when `on = true` the LED is set to 0 (which turns it off), so I believe that should be reversed.

Disclaimer: I only have a basic electronics understanding so I could be wiring things backwards. (I have a resistor pulling the negative lead to ground when the GPIO pin is activated.) But this change caused the expected results for me. :)

Thank you for the wonderful code! Homebridge has been too unreliable for me once I add more than one or two plugins, so I am excited to be able to create my own devices which can speak directly to Homekit!

brianhanifin, 2018-02-23:
Update: scratch that, actually my change makes the `1 : 0` redundant, so this is my final line of code for this.

`gpio_write(led_gpio, on);`

maximkulkin, 2018-02-23:
What kind of board do you have? My Makerfocus NodeMCU and HiLetgo NodeMCU boards behaved correctly with existing code.

brianhanifin, 2018-02-23:
I picked them up at amazon. I picked them kind of randomly. I was unable to find any documentation for them... they seem to me made by DOIT, but they don't provide any documentation for that model.

[Makerfocus 2pcs ESP8285 ESP-M2 CH340 Development Board WIFI Serial Port Module CH340 Compatible with ESP8266](https://www.amazon.com/dp/B075K48R1Q?ref=yo_pop_ma_swf)

After watching more YouTube videos I ordered some D1 Mini boards as well. They  will arrive this evening so I can try the code on it as a test.

I suppose if this is an inconsistent behavior the an INVERT_LED_STATUS flag could be set at the top of the example code just in case. I will report back later.

brianhanifin, 2018-02-24:
OK I just tested my LED code on the [Makerfocus (Wemos clone?) D1 Mini](https://smile.amazon.com/gp/product/B01N3P763C/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1), and using pin 2 controls the internal blue LED light. Its state is the inverse of the Makerfocuse ESP8285 board.

I guess the truth is, these are just code examples and if we aren't clever enough to correct an inverted LED state then we sure aren't going to be able to do anything else with this SDK. It probably isn't worth changing anything. ;)

closed 2018-02-24 by brianhanifin