maximkulkin/esp-homekit-demo #39: Proposal: Create Discussion Group for "esp-homekit"

brianhanifin, 2018-02-22:
Is there interest in creating a Facebook group (or Reddit)? That way the users could support each other while the developer(s) can spend more time focusing on bugs and improvments. If someone is unsure if something should be reported as a bug it could be discussed there first. 

It could also be a place to brag about the interesting and creative ways you have put this library to use! ;)

brianhanifin, 2018-02-22:
Here is a link to one I have started. Join if you want, or tell me this is a bad idea and why I should shut it down. 😀 

https://www.facebook.com/groups/2057393107867342/

renssies, 2018-02-22:
I know that some members might be concerned with privacy, they won't join a facebook group.

So I made a Subreddit: https://www.reddit.com/r/ESPHomeKit/. Reddit is also better at handling code blocks in comments and posts.

brianhanifin, 2018-02-22:
@renssies That's great! Thank you for doing that, I just joined up.

AdySan, 2018-02-22:
I’d also recommend gitter. Works great with GitHub logins.

brianhanifin, 2018-02-23:
@AdySan I used Gitter a few times when I contributed to an open source project a few years ago. If I remember correctly it is a nice chat interface for developers working on a project together. But a nice reddit is a nice because it becomes a searchable knowledgeable.

ageorgios, 2018-02-28:
+1 for Gitter

renssies, 2018-02-28:
I have created a gitter as well: https://gitter.im/ESP-HomeKit/Lobby

closed 2018-02-28 by maximkulkin