maximkulkin/esp-homekit-demo #40: Sonoff's GPIO14 as Wall Switch Button

osvaldoasn, 2018-02-23:
Excellent work with esp-homekit!

There's a way to use Sonoff's GPIO 14 with a regulat wall switch?
I've seen this on [https://github.com/arendst/Sonoff-Tasmota](url) and with many switch options, but it needs a HomeBridge.

Thanks.

maximkulkin, 2018-02-23:
Can you describe it in more detail? I thought [sonoff_basic](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/sonoff_basic) example covers that (except for it uses a different pin for button).

renssies, 2018-02-24:
You should be able to do it by connecting the switch to 3,3v and the extra header pin. Next, in the code, you make sure GPI014 is an input pin and you can check the state of the switch. 

I've not included this in my Sonoff basic example because these examples are really examples. They are meant for people to make there own implementation :)

renssies, 2018-02-24:
@maximkulkin If you've programmed a Sonoff basic you will have noticed that 1 pin is unused. That pin is connected to CPIO 14. The PCB for the Sonoff Basic and Sonoff TH is the same. But the Sonoff TH has a 2.5mm jack connected to GPIO14, making it possible to connect accessories.

RavenSystem, 2018-02-25:
@osvaldoasn , that functionality is supported in this firmware:
https://github.com/RavenSystem/esp-homekit-devices/tree/master/devices/Sonoff_Basic

You only must connect your wall-switch to GPIO14 and GND.

lzparkma, 2018-02-25:
I modified the Sonoff_Basic demo to include a toggle on GPIO14 a few weeks ago but have no idea how to push this back to this project or if it is desired. Is a not exactly a pretty hack but it works. Please advice if this example is desired.

osvaldoasn, 2018-02-26:
@lzparkma please include it to the project excample and/or send to me at osvaldo@dw8.com.br

Thanks.

lzparkma, 2018-02-26:
@osvaldoasn and @maximkulkin have created pull request from my fork. Would strongly suggest someone who knows what they are doing check this code as is was more trial-and-error than by design.

closed 2018-02-28 by osvaldoasn