maximkulkin/esp-homekit-demo #41: Add: An Example to control Ws2812b / SK6812 RGB(W) Stripes

dbambus, 2018-02-24:
Hey maximkulkin,

you really did an amazing job creating all those examples for HomeKit on an ESP8266. I used it to create a bunch of different accessories, but I was wondering how to control RGB(W) LED Stripes using HomeKit. I took a look in your Light example, but couldn't work it out.

David

renssies, 2018-02-24:
The lightbulb example uses regular LEDs, controlling them with transistors and PWM.
WS281x LEDs are digital LEDs, you can control them individually from one data wire.

To find an example of how to control WS281x LEDs using `esp-open-rtos` (what the homekit code is based on), have a look at this: https://github.com/SuperHouse/esp-open-rtos/tree/master/examples/ws2812_i2s and the fireplace example. Both use i2s to control WS281x LEDs. 
 
Controlling the WS281x over i2s is required because of the timing that is incorrect because of a lot of network traffic.

maximkulkin, 2018-02-24:
Take a look at “fireplace” example.

closed 2018-03-05 by maximkulkin