maximkulkin/esp-homekit-demo #44: Automations not working reliable

Arduingo, 2018-02-25:
I have been testing the examples with automations and found a strange behavior.

My setup Sonoff with the example compiled with the latest build in the page only change mDND buffer increased to 1.5k, WEMO D1 with the Led example, HUE lamps

Create 2 automations were one device is the On/Off master and the others are On/Off slave so if master is turned ON slaves should turn on, if master is turned Off slaves are turned Off

Steps to reproduce:

Power up both devices and wait for the connection and open the homeApp
turn on master -> slaves will turn on, wait 5 seconds
turn off master -> slaves will turn off, wait 5 seconds
turn on master -> Slaves will stay off
Wait 1-2 min
Turn on master -> Slaves will turn on

I also tested the sonoff as master and HUE as Slave and the result is the same

Using a HUE lamp as master and the Sonoff and Led example as slave, works as expected

Using a HUE lamp as master and another HUE lamp as Slave works as expected

Wondering if its something only happening to me or something common to other makers

maximkulkin, 2018-02-26:
Who is who in your repro? Are both master and slave an accessories powered by esp-homekit? Can you provide logs (with HOMEKIT_DEBUG=1)?

Arduingo, 2018-02-26:
The Sonoff is the Master, HUE and WemoD1 flashed with LED example are the slaves
Also tested using only  two HUE lamps one master one Slave to validate correct behavior
WemoD1 and Sonoff uses Esp-homekit framework examples compiled with the latest release from this GIT

Attached are 2 Logs with some inline comments:

Log1.txt: Boot after sonoff reflash, 1st Try it worked, 2nd Try automation failed
log2.txt: Reset by removing power and reconnect, 1st try Fail, 2nd try Fail, wait 1 min Worked, retry fail, wait 1 min worked, retry fail

Both logs are commented with separators (====) between success and fail events

Please let me know if you need something else

[LOG1.txt](https://github.com/maximkulkin/esp-homekit-demo/files/1758949/LOG1.txt)
[LOG2.txt](https://github.com/maximkulkin/esp-homekit-demo/files/1758950/LOG2.txt)

Arduingo, 2018-03-03:
After some experiments I removed the Accessory from the HomeApp then erase flash, reflash a fresh build, added again the accessory and now everything is working as expected.

closed 2018-03-03 by Arduingo