maximkulkin/esp-homekit-demo #49: Error upload firmware

beastgrim, 2018-02-28:
Hello!

I have a trouble with flashing to Wemos D1 mini pro device.
I followed instructions and compiled "exemples/led" projet but can't flash to device.

My trace log:

_MacBook-Pro:esp-homekit-demo admin$ make -C examples/led test
esptool.py -p /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fs 32m -fm qio -ff 40m \...
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.3
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8285
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2637, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2630, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2360, in main
    print("Features: %s" % ", ".join(esp.get_chip_features()))
  File "/usr/local/bin/esptool.py", line 897, in get_chip_features
    features += ["Embedded Flash"]
TypeError: cannot concatenate 'str' and 'list' objects
make: *** [flash] Error 1_

Can you help me? Thanks for any help!

maximkulkin, 2018-02-28:
Update esptool to version 2.3.1, version 2.3 has this bug:
```
pip install -U esptool
```

beastgrim, 2018-02-28:
Thank you very much! Everything is working!

closed 2018-02-28 by beastgrim