maximkulkin/esp-homekit-demo #51: Detailed instructions on how to do this

eneski, 2018-03-13:
Hello,
I am kinda new to all this and I am currently using my WeMos D1 mini in connection with Homebridge. It would be very cool, if it was possible to do all this without Homebridge, but even if you have instructions, I still can not manage to get this on my board. I would very appreciate it if anyone had detailed step by step instructions on how to do this ! :)

RavenSystem, 2018-03-14:
@eneski , I think that homekit/include is well documented, and you have a lot of working examples in this repo.

This HomeKit API is very easy to use, but maybe you are talking about coding with [esp-open-rtos SDK](https://github.com/SuperHouse/esp-open-rtos) for your device.

eneski, 2018-03-14:
the problem is not the documentation .. the problem is that I don‘t even know where to enter those commands. I got a Windows PC and a Mac, do i enter that at the command line or terminal or something?

ageorgios, 2018-03-15:
Prefer a Linux machine (like Ubuntu) on a VM, I had success with Ubuntu on Parallels on Mac

Arduingo, 2018-03-15:
Follow These two video Tutorials, they will teach you how to setup a dev environment in a Linux Virtual Machine using free VirtualBox, After that, you will have a place were to enter all the commands, run the demos and don't mess with your system.

https://www.youtube.com/watch?v=kYEzEFH5LiM 
https://www.youtube.com/watch?v=T-oSjMCmNYk

Additional to what is installed in the videos, you will need to clone and compile Free-Rtos and ESP-Homekit by following the instructions in the repos

faronov, 2018-03-15:
don't understand your problem if you have a mac
detailed description have all involved project
1. you need install esp-open-sdk
open https://github.com/pfalcon/esp-open-sdk#macos
and you will see detailed instruction for macOS also
2. checkout esp-open-rtos (instruction here https://github.com/SuperHouse/esp-open-rtos)
if simple just run 
this can be inside /Volumes/case-sensitive folder
git clone --recursive https://github.com/Superhouse/esp-open-rtos.git
run
SDK_PATH=/Volumes/case-sensitive/esp-open-rtos
3. set the correct path for your serial device
export ESPPORT=/dev/tty.SLAB_USBtoUART
you can found correct device run /dev/tty.

Dave1001, 2018-03-25:
Hi - I'm also using a Wemos D1 mini on a Linux box (Ubuntu) - was switching between examples and occasionally getting distracted so wrote a bash script to handle setting the environment variables, build and flash. 
See pull request #54 - it might make life a little easier...

maximkulkin, 2018-03-26:
I wrote in PR but will duplicate it here.

I personally do it like this:

1. Install autoenv:
    ```shell
    brew install autoenv
    ```
2. In my project directory create .env file:
    ```shell
    export SDK_PATH=~/Projects/eps8266/esp-open-rtos
    export ESPPORT=/dev/tty.SLAB_USBtoUART
    export FLASH_SIZE=32
    export HOMEKIT_DEBUG=1
    ```
Autoenv extension modifies your shell to source .env file whenever you enter directory with that file or any subdirectory.

Then, you do not actually need to do `make -C <dir> all`, you can just do `cd <dir>` and then you can do simple commands like `make all`, `make test` or `make monitor`. E.g.
```shell
cd ~/Projects/esp8266/esp-homekit-demo/examples/led
make all
make erase_flash
make test
```
Subsequent changes can be flashed with only one command:
```shell
make test
```

maccacino, 2018-04-14:
Arghghghghhg.... When adding your PATH variables, make sure they are separated with a ":"... 

/some/path/to/variable:/other/path/to/variable.

Thus, when editing the bashrc file with;
`$ nano ~/.bashrc`

make sure to add the code like this;
`export PATH=$PATH:/home/usr/esp-open-sdk/xtensa-lx106-elf/bin`

and not like this
`export PATH=$PATH/home/usr/esp-open-sdk/xtensa-lx106-elf/bin`

maximkulkin, 2018-04-17:
I'm sorry to hear that you had problems figuring that out.

I want to articulate that: this project is a framework for developers to develop firmwares. That means that it's a DIY kit, not a finished product that works out of the box. As author I do not have neither resources, nor intent to invest into making this a product. I (selectively) accept patches that add more examples, but I do not plan to have a comprehensive set of examples for each and every ESP8266 enabled product on the market.

There are other developers that might fill the gap. E.g. @RavenSystem has a [repository](https://github.com/RavenSystem/esp-homekit-devices) with more Sonoff firmwares. You should probably check those.

closed 2018-05-09 by maximkulkin