maximkulkin/esp-homekit-demo #52: Is it possible to edit it?  ”sonoff_basic_pwm” —> ”2-gang pwm in same chip”

jukorp, 2018-03-18:
Button 1 -> pwm1 up
Button 2 -> pwm1 down
Button 3 -> pwm2 up
Button 4 -> pwm2 down
Status led1 -> pwm1 on
Status led2 -> pwm2 on

Is it difficult to edit the code?

Arduingo, 2018-03-22:
I guess you mean use the current HW to handle 2 switch and 2 leds. Currently the Sonoff only have one spare I/O pin available with easy access in the header, this I/O pin can be used for a push button or a led. You may use de RX on the serial port for an additional button but will lost the capability to debug.

jukorp, 2018-03-22:
I'm using the esp-12E module.
There are several gbio available.
I'm designing and manufacturing hardware myself.

this is datasheet module:
http://www.kloppenborg.net/images/blog/esp8266/esp8266-esp12e-specs.pdf
[image]

Arduingo, 2018-03-22:
Then just its just a matter of use the right I/O pin, is quite easy add a button and a led to the current code. ESP-12 is based on the ESP8266 chip, the code will work without changes.

Even if its a single HW, Homekit you need to create an accessory with 2 services, the current example have 1 light service in main.c:

HOMEKIT_SERVICE(LIGHTBULB, .primary=true,
            .characteristics=(homekit_characteristic_t*[]){
                HOMEKIT_CHARACTERISTIC(NAME, "Sonoff Dimmer"),
                &lightbulb_on,
                HOMEKIT_CHARACTERISTIC(BRIGHTNESS, 100, .getter=light_bri_get, .setter=light_bri_set),
            NULL
        }),

Add one more and you will have 2 lights, add the getter and setter for that light, basically you will need to clone the code and make some modifications.

lzparkma, 2018-04-03:
I have created an example for the Sonoff Dual R2 (sonoff_dual_toggle) which has two relays and implemented as two homekit switch characteristics. From this and the sonoff_basic_pwm example (or other pwm example) you should be able to work something out.

closed 2018-05-09 by maximkulkin