maximkulkin/esp-homekit-demo #55: Simple dev tooling on a Mac

cottsak, 2018-03-26:
Hey folks,

I'm new to using the esp SDK and building from the command line.

I've just got the exmples/led working after a couple of hours of messing around with repos, building, make errors and environment vars. I can see how https://github.com/maximkulkin/esp-homekit-demo/pull/54 might help.

Anyway.. what's a really simple dev workflow/toolset for improving on the simple examples? For example, I can use vscode and the terminal to build but I don't have some of the "built in" features of the Arduino IDE.
* What's the best approach to syntax/api validation?
* Where can I get basic help on the differences between `setup()` and `loop()` as compared to [how this works with the SDK](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/led/led.c)?
* what's the solution to serial monitoring when building and flashing from the command line?

Are most of the Arduino compatible libs going to "just work" using the SDK?
I'm keen to augment a simple relay implementation of example/led with the [WiFiManager](https://github.com/tzapu/WiFiManager) and [DoubleResetDetector](https://github.com/datacute/DoubleResetDetector). Is this going to be straightforward?

maximkulkin, 2018-03-26:
> I've just got the exmples/led working after a couple of hours of messing around with repos, building, make errors and environment vars.

Have you tried following instructions on front page? What problems have you encountered?

For syntax validation you just use whatever you favorite editor is. For API validation I guess you can set up include paths in your C/C++ capable IDE to be able to find all header files (including ESP_OPEN_RTOS ones).

For SDK differences you'd better consult SDK documentation.

For serial monitoring and flashing you should use either `make test` or `make flash` + `make monitor`.

Arduino libs generally won't work at all. You need ones that were specially designed for ESP_OPEN_SDK.

Dave1001, 2018-03-26:
Thanks Maxim - it looks like cottsak and I are following the same journey ;) That said - its all up and running for me so very grateful for your work!

cottsak - to add to Maxim's comments - the libraries from Arduino are sort-of-C++ so would be a fairly long shot to write C wrappers and link object code with esp-open-rtos (TLDR - way too hard). That said - I've found there's some reasonably good examples at "\your\path\to\esp-open-rtos\examples" that cover the main interfaces. If its just a simple relay you need - just hook the relay up to the LED pin! (depending on your hardware, you might need a level converter to bump the pin up to 5V TTL - there's plenty on ebay)

cottsak, 2018-03-27:
Thanks folks. Yes, I gather that if I want the specific behaviours from those Arduino libs I mentioned, I'll have to reimplement them myself.

closed 2018-05-09 by maximkulkin