maximkulkin/esp-homekit-demo #58: Licensing for commercialisation

cottsak, 2018-03-27:
It seems that the ESP SDK is complex enough, let alone this open source HomeKit collection of dependancies. Can anyone help me figure out if a fork of this codebase can be legally packaged into an off-the-shelf product (eg. [Sonoff Basic](http://sonoff.itead.cc/en/products/sonoff/sonoff-basic), etc) and then sold?

maximkulkin, 2018-03-27:
esp-homekit is published under MIT license and ESP_OPEN_RTOS under BDS 3-clause. So, both are permissive.

renssies, 2018-03-27:
You will get into legal trouble if you sell the Sonoff Basic with the esp-homekit based firmware commercially:

1. The esp-homekit code is based on the non-commercial specification of HomeKit and while using it you agree to the terms that you are not allowed to sell this commercially without entering the Apple MFI program. 
2. Modifying and reselling the Sonoff will get you into legal trouble with iTead

cottsak, 2018-04-05:
@renssies No I don't mean reselling Sonoff Basic. I meant selling something *like* the Sonoff Basic. Sorry, I should have been more clear.

If I developed my own physical product, similar to the Sonoff Basic (in the sense that it was a simple mains-powered wifi switch), would I encounter complexities in the open source dependency chain with regard to licences that don't permit commercial use?

renssies, 2018-04-05:
@cottsak I'm no expert on this, but yes you can use it for a commercial product. However, you do need to give copyright notice. Most of the times this will happen in-app or in the manual. 

However, if you are making a commercial physical product I suggest entering the MFI program, several manufacturers (including Expressif and the ESP32) have an SDK available for MFI members :)

closed 2018-05-09 by maximkulkin