maximkulkin/esp-homekit-demo #60: How can I see the change in the app when a local button changes an output on the ESP

korstiaanS, 2018-04-20:
Hi all,

First of all GREAT work!
I have got it up and running by starting with the LED example.
I then changed it to 2 services instead of 1 so that only 1 ESP8266 controls 2 outputs.
This al works fine, in the Home app on the iPhone I see 2 devices that comes from 1 ESP.
But I will control the outputs (lights) also locally via buttons connected to the ESP.
I do NOT want to know the state of these buttons but if I turn an output via a button ON or OFF, the light goes ON or OFF and I want to see this change in the app as well.
How do I do this?
I suppose I need to send a notify/event every time a light changes from state (via the hardware button).
I'm not so familiar with the calls, so how do I implement this?

Can somebody help me to to get this functionallity inside my program?
 
Korstiaan

maximkulkin, 2018-04-20:
I've just did this for my floor lamp.

[image]

The floor lamp has two lights (see screenshot, although this is not exactly my lamp) and a knob to switch between the two. So I've rewired it to three 18 awg wires and two thin (signal) wires going from bottom (where control board is) to mid point (where knob is). I've used Sonoff Dual R2 board. My knob has 4 states: both off, top on, bottom on, both on. So I've found out if I connect signal wires to common and top wires of knob, when I will rotate the knob it will switch 0, 1, 0, 1. So the firmware just detects level change (either 0->1 or 1->0) and treats it as signal.

https://gist.github.com/maximkulkin/82834ea47ce2e116e4188e61aaca6042

korstiaanS, 2018-04-20:
Hi Maxim,

Great example!
If you open the app on the iPhone and leave it open, and then if you change a lamp state via your knob does the app sees this change? So, without leaving or refreshing the app, you see the change in the app as well without any interaction/manipulation on the phone?

maximkulkin, 2018-04-20:
yes

korstiaanS, 2018-04-20:
Ok, that's what I'm looking for!
I will try to understand/experiment your code later on this weekend.
If I take a quick look you changed in the service from a .getter and .setter to a pointer to a homekit_characteristic_t.
But there I see the setter but no getter?
So I don't see the flow how the app gets the status (for example when opening the app the first time) when there is no getter?
Does that mean that the current state of a lamp is only controlled via the notifications (homekit_characteristic_notify(&top_light_on, top_light_on.value))?

maximkulkin, 2018-04-20:
So thing with characteristic: every characteristic stores value internally. If you define getter or setter (or both) those are used to get that value by framework. If any of that is missing, it just uses default way to get value (which is read/write directly from characteristic). You can see that after defining my own setter, I still set it in characteristic manually (e.g. `top_light_on.value = HOMEKIT_BOOL(top_on);`)

There is a "toggle" attached to a button pin: it triggers callback whenever logical level on that pin changes. Callback advances current lamp state to next one, reconfigures relays, updates characteristics and send characteristic notifications (that's how your iOS device is notified that characteristic got changed).

korstiaanS, 2018-04-22:
Hi,

I sorted it out and it worked also for me now thanks to your example (the floor lamp).

I'm extending my program and suddenly it stopped working (I cannot add it to my Home, it is not visible anymore)?
I realise now that I have an ESP8266 with 2MB (16Mbit). [Link to Olimex](https://www.olimex.com/Products/IoT/MOD-WIFI-ESP8266-DEV/open-source-hardware)
Until now I used the default settings (for 4MB, 32Mbit) and this worked.
Maybe this is the reason why my program stopped working correctly?
Do you know the correct setting for a 2MB?
I think this is:
export FLASH_SIZE=16
but what for 
export HOMEKIT_SPI_FLASH_BASE_ADDR=?

Korstiaan

closed 2018-05-09 by maximkulkin