maximkulkin/esp-homekit-demo #61: Is building on Docker possible?

pbendersky, 2018-04-21:
Just saw this comment https://github.com/pfalcon/esp-open-sdk/issues/292#issuecomment-340453970 by @maximkulkin and wondered if it was related to this project... If so: were you able to build esp-homekit using Docker? Would you recommend that approach?

maximkulkin, 2018-04-21:
I personally have SDK packaged into Docker container, but that thing has its own issues that’s why I do not actively promote this method. Main issue is that you can not flash firmware from inside container, so that part I do from host system. That is also why build instructions look like ‘make -C examples/foo’ instead of ‘cd examples/foo ; make’

pbendersky, 2018-04-21:
I think it’s a clean approach. Would you mind sharing your Dockerfile? If I get it running I can send a PR documenting it.

mccahan, 2018-05-05:
Not Docker, but a while ago I managed to create a VirtualBox Vagrant environment (https://github.com/mccahan/ESP-HomeKit-Vagrant-BuildEnv), which allows you to share the USB device with the VM and flash from within the VM itself.

pbendersky, 2018-05-07:
Thank you @mccahan. I was able to build just fine with the Dockerfile @maximkulkin posted here: https://github.com/pfalcon/esp-open-sdk/issues/292#issuecomment-340453970

The only thing is that you need to make using `docker run`, but then flash locally from the Mac.

So, you end up with something like:
`docker run -v `pwd`:/esp -v ~/Projects/esp/esp-open-rtos:/sdk -w /esp -e SDK_PATH=/sdk esp-sdk:latest make -C examples/led all` to build a sample, and then `make -C examples/led test` to actually flash it.

I'll try to document it and send a PR here, so others can use this approach.

maximkulkin, 2018-05-09:
@pbendersky I personally have following bash helper function:
```shell
docker-run() {
    docker run -it --rm -w /project -v `pwd`:/project "$@"
}
```
It launches a temporary container and mounts current directory into `/project` directory inside container. So my build run looks like this: `docker-run esp-rtos make -C examples/foo all`

closed 2018-05-09 by maximkulkin