maximkulkin/esp-homekit-demo #62: make -C /example/led all error

xfce, 2018-04-23:
hi all,
   i setup as the readme, when make -C /example/led all , got the error bellow, could someone give any advices ,thanks

$ make -C ./examples/led all
make: Entering directory '/home/bi4wms/projects/esp-open-sdk/ESP8266_RTOS_SDK/esp-homekit-demo/examples/led'
CC /home/bi4wms/projects/esp-open-sdk/esp-open-rtos/open_esplibs/libmain/user_interface.c
/home/bi4wms/projects/esp-open-sdk/esp-open-rtos/open_esplibs/libmain/user_interface.c:11:23: fatal error: lwip/dhcp.h: No such file or directory
 #include "lwip/dhcp.h"
                       ^
compilation terminated.
/home/bi4wms/projects/esp-open-sdk/esp-open-rtos//open_esplibs/component.mk:15: recipe for target 'build/open_esplibs_libmain//user_interface.o' failed
make: *** [build/open_esplibs_libmain//user_interface.o] Error 1
make: Leaving directory '/home/bi4wms/projects/esp-open-sdk/ESP8266_RTOS_SDK/esp-homekit-demo/examples/led'

maximkulkin, 2018-04-23:
That looks like if you did not sep SDK_PATH variable pointing to ESP_OPEN_RTOS directory

closed 2018-04-27 by maximkulkin

likewise, 2018-12-12:
That, or you didn't checkout esp-open-rtos recursively, so you are missing the submodules.
cd esp-open-rtos
git submodule update --init --recursive