maximkulkin/esp-homekit-demo #65: Debug output keeps sending data when I don't want to

korstiaanS, 2018-04-26:
Hi,

I noticed that, even when HOMEKIT_DEBUG=0 the debug messages keep coming out via the serial output. (And I want the GPIO/TXT output for something else).
Is it because 
`#ifdef HOMEKIT_DEBUG`
is used instead of 
`#if HOMEKIT_DEBUG > 0
` 
Also when I don't have a HOMEKIT_DEBUG environment variable I still get the output?

maximkulkin, 2018-04-26:
Its a build time variable. When you change it, you need to rebuild the project (`make rebuild`)

korstiaanS, 2018-04-26:
Hi Maxim,

Yes of course! It works now. Thanks.

PS:
I also had some problems because I'm using XCODE as an development environment. I can build there to see if I don't have any faults and to bring me to the error (this compiles to a separate directory of Xcode). But if I press Run in Xcode (=erase_flash + flash + start the serial monitor coolterm all in 1 click full automatically) from xcode I'm using (via an own made helper app) another script and this compiles in the build subdirectory of the project.

korstiaanS, 2018-04-26:
Tested it all in a terminal:

env shows HOMEKIT_DEBUG=0

First I did make -C myHomekitOnEsp rebuild.
Then 
make -C myHomekitOnEsp erase_flash flash
Still I get this output (part of it is displayed):

```
dhcp client start...
>>> HomeKit: Generated new accessory key
>>> HomeKit: Waiting for IP
ip:192.168.56.136,mask:255.255.255.0,gw:192.168.56.254
>>> HomeKit: Got IP, starting
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
```

pcsaito, 2018-04-26:
What is expected if you never set the HOMEKIT_DEBUG?

Calling "env" does not shows the HOMEKIT_DEBUG at all, but generate the debug logs korstiaanS comment above.

korstiaanS, 2018-04-26:
If I rebuild with no HOMEKIT_DEBUG at all I get the same output as HOMEKIT_DEBUG=0:

```
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 3D:27:91:98:A0:22
>>> HomeKit: Waiting for IP
scandone
add 0
aid 17
cnt 

connected with Dekenij56, channel 7
dhcp client start...
ip:192.168.56.136,mask:255.255.255.0,gw:192.168.56.254
>>> HomeKit: Got IP, starting
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 5] Pair Setup Step 1/3
>>> HomeKit: [Client 5] Pair Setup Step 2/3
>>> HomeKit: [Client 5] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 225E3169-4F32-4FBA-8710-E16FF18C61D3
>>> HomeKit: Configuring mDNS
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: [Client 5] Successfully paired
>>> HomeKit: [Client 5] Closing client connection
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 225E3169-4F32-4FBA-8710-E16FF18C61D3
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with 934500CC-9898-4372-82B8-020DB6820A6B
>>> HomeKit: Got new client connection: 6
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 7
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 7] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with 225E3169-4F32-4FBA-8710-E16FF18C61D3
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 7] Pair Verify Step 2/2
>>> HomeKit: [Client 7] Found pairing with 225E3169-4F32-4FBA-8710-E16FF18C61D3
>>> HomeKit: [Client 7] Verification successful, secure session established
>>> HomeKit: Got new client connection: 8
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 6] Get Accessories
>>> HomeKit: [Client 7] Closing client connection
>>> HomeKit: [Client 8] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 6] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 8] Pair Verify Step 2/2
>>> HomeKit: [Client 8] Found pairing with 225E3169-4F32-4FBA-8710-E16FF18C61D3
>>> HomeKit: [Client 8] Verification successful, secure session established
>>> HomeKit: [Client 8] Get Accessories
```
If I build with HOMEKIT_DEBUG=1 I get more detail (only a part of it below, there too much):

```
...
>>> client_send: [Client 6] Sending payload: 0\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 6] Finished processing
>>> homekit_client_process: [Client 5] Got 288 incomming data
>>> homekit_client_process: [Client 5] Decrypting data
>>> homekit_client_process: [Client 5] Decrypted 270 bytes, available 0
Decrypted data (270 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: my2digital\\032ESP's._hap._tcp.local\x0D\x0AContent-Length: 137\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"ev":true},{"aid":1,"iid":11,"ev":true},{"aid":1,"iid":14,"ev":true},{"aid":1,"iid":15,"ev":true}]}"
>>> HomeKit: [Client 5] Update Characteristics
>>> homekit_server_on_update_characteristics: Free heap: 26504
>>> homekit_server_on_update_characteristics: [Client 5] Processing element {
."aid":.1,
."iid":.10,
."ev":.true
}
>>> homekit_server_on_update_characteristics: [Client 5] Processing element {
."aid":.1,
."iid":.11,
."ev":.true
}
>>> homekit_server_on_update_characteristics: [Client 5] Processing element {
."aid":.1,
."iid":.14,
."ev":.true
}
>>> homekit_server_on_update_characteristics: [Client 5] Processing element {
."aid":.1,
."iid":.15,
."ev":.true
}
>>> homekit_server_on_update_characteristics: [Client 5] There were no processing errors, sending No Content response
>>> client_send: [Client 5] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 5] Finished processing
>>> homekit_client_process: [Client 7] Got 170 incomming data
>>> homekit_server_on_pair_verify: HomeKit Pair Verify
>>> homekit_server_on_pair_verify: Free heap: 25308
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (32 bytes):
...
```

Anyway I can't configure it without output.

maximkulkin, 2018-04-26:
There is normal output that is always printed and then there is a debug messages that give you more insight on what is happening.

I guess, you can just skip initializing uart.

closed 2019-01-27 by maximkulkin