maximkulkin/esp-homekit-demo #66: Is using a static IP possible?

korstiaanS, 2018-04-27:
Hi,

Can you use a static IP instead of DHCP?
I already looked at the esp-open-rtos site but I don't find anything that works.
Can somebody help me to set this up?
Thanks.
Korstiaan

maximkulkin, 2018-04-27:
Haven't tried that, but it looks like you need to do the following:
```c
#include <espressif/user_interface.h>

sdk_wifi_station_dhcpc_stop();

static struct ip_info my_ip = {
    .ip = IPADDR4_INIT_BYTES(192, 168, 0, 55),
    .netmask = IPADDR4_INIT_BYTES(255, 255, 255, 0),
    .gw = IPADDR4_INIT_BYTES(192, 168, 0, 1)
};
sdk_wifi_set_ip_info(STATION_IF, &my_ip);
```

korstiaanS, 2018-04-28:
Hi,

No success.
I'm also asking at esp-open-rtos, see [https://github.com/SuperHouse/esp-open-rtos/issues/619](url)

closed 2019-01-27 by maximkulkin