maximkulkin/esp-homekit-demo #69: Error

Kristian8606, 2018-04-30:
esptool.py -p /dev/tty.wchusbserial1420 --baud 115200 write_flash -fs 16m -fm qio -ff 40m 0x0 ../../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./f
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size {detect,4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                           [--no-progress] [--verify]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '../../bootloader/firmware_prebuilt/rboot.bin'

how to flash ?

maximkulkin, 2018-04-30:
You probably specified SDK_PATH as relative path. Try using absolute path instead.

Kristian8606, 2018-04-30:
Please make your installation instructions, flash, and all the rest.
I try for a week and it does not work.

Kristian8606, 2018-04-30:
I need a comprehensive step-by-step guide to succeed.