maximkulkin/esp-homekit-demo #70: Random AC interference on sonoff_basic_toggle

pcsaito, 2018-05-03:
Sometimes my kitchens lights get toggled when I turn on some electrical appliances. It doesn't happen every time, but when it happens it was in perfect sync with the mixer or the fridge compressor starting.

I'm testing a solution to debounce based a low pass filter on the switch input to reject noise and using the absolute value of the switch to decides the toggle.

Anyone are having these behaviors too?

pcsaito, 2018-05-03:
Thanks for the capacitor advice! I have read that a in series resistor could help too. If the problem continues I will try both.

But for now looks like the LPF solved the issue, but I need more testing because it’s random for me.

Would you like to try? You problem appears to be more reproducible.

https://github.com/pcsaito/esp-homekit-demo/tree/LPFToggle

pcsaito, 2018-05-10:
@maximkulkin My software fix for this problem is already working for 7 days without any false trigger or stability issue. 
With the current sonoff_basic_toggle demo I had at least some false triggers every couple of days, there are more days with the problem than without. 

Should I open a PR?

maximkulkin, 2018-05-10:
Sure

lzparkma, 2018-05-15:
FYI have also been testing this change and agree is more stable.

PR #76 (2018-05-23): Sonoff basic toggle with low pass filtering  
This PR implements a low pass filter on gpio value instead of a time debounce to toggle.

With the default parameters it will filter up to 6 wrong readings and introduces about 60ms delay.

Fixes #70
closed 2018-05-23 by maximkulkin