maximkulkin/esp-homekit-demo #71: Failed to clone 'components/WS2812FX'

streicherlouw, 2018-05-04:
I am attempting to check out the new led strip animation code, but getting a failure from git. I suspect that there was a problem with the branching or naming, as the submodule update looks for host "github.com (192.30.255.113)", which seems to concatenate a local network address with the hostname. My internal network is all int he 10.x address space, so I am pretty sure it's not a local problem.

Streichers-15-MBP:case-sensitive streicher$ git clone https://github.com/maximkulkin/esp-homekit-demo.git
Cloning into 'esp-homekit-demo'...
remote: Counting objects: 256, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 256 (delta 3), reused 10 (delta 1), pack-reused 242
Receiving objects: 100% (256/256), 1.15 MiB | 865.00 KiB/s, done.
Resolving deltas: 100% (127/127), done.
Streichers-15-MBP:case-sensitive streicher$ cd esp-homekit-demo/
Streichers-15-MBP:esp-homekit-demo streicher$ git submodule update --init --recursive
Submodule 'components/WS2812FX' (git@github.com:pcsaito/WS2812FX-rtos.git) registered for path 'components/WS2812FX'
Submodule 'components/cJSON' (https://github.com/maximkulkin/esp-cjson) registered for path 'components/cJSON'
Submodule 'components/homekit' (https://github.com/maximkulkin/esp-homekit) registered for path 'components/homekit'
Submodule 'components/wifi_config' (https://github.com/maximkulkin/esp-wifi-config) registered for path 'components/wifi_config'
Submodule 'components/wolfssl' (https://github.com/maximkulkin/esp-wolfssl) registered for path 'components/wolfssl'
Cloning into '/Volumes/case-sensitive/esp-homekit-demo/components/WS2812FX'...
The authenticity of host 'github.com (192.30.255.113)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'github.com,192.30.255.113' (RSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:pcsaito/WS2812FX-rtos.git' into submodule path '/Volumes/case-sensitive/esp-homekit-demo/components/WS2812FX' failed
Failed to clone 'components/WS2812FX'. Retry scheduled
Cloning into '/Volumes/case-sensitive/esp-homekit-demo/components/cJSON'...
Cloning into '/Volumes/case-sensitive/esp-homekit-demo/components/homekit'...
Cloning into '/Volumes/case-sensitive/esp-homekit-demo/components/wifi_config'...
Cloning into '/Volumes/case-sensitive/esp-homekit-demo/components/wolfssl'...
Cloning into '/Volumes/case-sensitive/esp-homekit-demo/components/WS2812FX'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:pcsaito/WS2812FX-rtos.git' into submodule path '/Volumes/case-sensitive/esp-homekit-demo/components/WS2812FX' failed
Failed to clone 'components/WS2812FX' a second time, aborting
Streichers-15-MBP:esp-homekit-demo streicher$

streicherlouw, 2018-05-04:
I think the problem might be on the last line of file .gitmodules
When I try to clone https://github.com/pcsaito/WS2812FX-rtos.git in stead of git@github.com:pcsaito/WS2812FX-rtos.git it works fine...

Streichers-15-MBP:esp-homekit-demo streicher$ cat .gitmodules 
[submodule "components/cJSON"]
	path = components/cJSON
	url = https://github.com/maximkulkin/esp-cjson
[submodule "components/wolfssl"]
	path = components/wolfssl
	url = https://github.com/maximkulkin/esp-wolfssl
[submodule "components/homekit"]
	path = components/homekit
	url = https://github.com/maximkulkin/esp-homekit
[submodule "components/wifi_config"]
	path = components/wifi_config
	url = https://github.com/maximkulkin/esp-wifi-config
[submodule "components/WS2812FX"]
	path = components/WS2812FX
	url = git@github.com:pcsaito/WS2812FX-rtos.git

RavenSystem, 2018-05-04:
Yes, issue is components/WS2812FX.

You can do:
`git submodule update --init --recursive components/cJSON components/homekit components/wifi_config components/wolfssl`
and of course, don't try to build any firmware that uses WS2812FX .

pcsaito, 2018-05-04:
I was not aware that using ssh instead of the https repo address could generate this issue.

I will submit a fix to that, but for now you can manually checkout the repository and copy to components folder