maximkulkin/esp-homekit-demo #73: Question about thermostat

Kristian8606, 2018-05-04:
DHT22 will work with the thermostat?
And why, when I set the thermostat off, does the relay turn on?

maximkulkin, 2018-05-05:
DHT22 should be supported.

For relay: you might have a different kind of relay which works in reverse.

Kristian8606, 2018-05-05:
DHT_TYPE_DHT11 I replaced it with DHT_TYPE_DXT22 and the sensor worked perfectly.
The question remains how to turn the function of the relay from the code itself?

pcsaito, 2018-05-05:
Just find the respective gpio_write call and reverse the second argument.

Example:
Change from
```
void heaterOn() {
    gpio_write(HEATER_PIN, false);
}
```

To
```
void heaterOn() {
    gpio_write(HEATER_PIN, true);
}
```

Don’t forget to reverse the “off” function as well.

Kristian8606, 2018-05-05:
Thanks for the reply.
Now it works when I set the temperature higher than the current's, but there is another.
When the thermostat is plugged into the socket, the relay works and when I set the heater switches off and starts functioning as I want.

I mean, when I plug it in, I do not want it plugged in until I plug it in from the app.

I would be grateful to you if and with that's help.

maximkulkin, 2018-05-09:
Obviously, you had to update initialization code with a reversed values as well. I have updated thermostat example to reuse `*On()` and `*Off()` functions so you only need to change it in one place. You can just rebase to latest code and it should work.

Closing issue. Feel free to reopen if you have any more questions.

closed 2018-05-09 by maximkulkin