maximkulkin/esp-homekit-demo #75: http accessory

supersjimmie, 2018-05-11:
Would it be possible to have something like an http accessory?
Like such: https://www.npmjs.com/package/homebridge-httpeverything

supersjimmie, 2018-05-15:
To be a bit more precise, what I am looking for:
I have an esp8266 (nodemcu) running a lot of home-automation stuff, like controlling my light dimmers, logging energy-usage (smart meter and solar), humidity, controlling the daikin airco, it has sensors for co2, light, temperature, and many more things.
This whole is now controlled by a simple web interface, running on that same esp2866.

What I am looking for is a homekit "frontend" that can do simple API-style calls to my existing esp8266. For example have a dimmer control, set the temperature and read values like co2, temp, humidity.
So at the homekit it is an accessory that just does simple web calls like 
http://1.2.3.4/light/up
http://1.2.3.4/aircotemp/21
http://1.2.3.4/read/co2

For that, I think I need some kind of accessory that accepts what I want to define, and then does a web request when I select it.

renssies, 2018-05-15:
Why do you want to build this with esp-homekit? Why not put Homebridge on a Raspberry Pi Zero (W)?

supersjimmie, 2018-05-15:
I like esp8266 more, especially because this could be running on an esp-01:
- Uses a lot less power  (24x7 saves many kwh's per year)
- Smaller (factor 10?)
- Cheaper (raspberry $10, esp-01 $1)
- I have several esp's laying around "waiting" to be used. :)

Since this esp-homekit-demo has nearly all right now, just except the "interfacing" part, why not expand this?  It has the part that communicates with apple, it has a few examples that use hardware, it has defined switches, thermostat, and many others working already.

supersjimmie, 2018-05-16:
I found an "http client" example here:
https://www.espressif.com/en/support/explore/sample-codes
So that's the part for doing the http requests.

The other part would then be to combine things like:
[button demo](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/button/main.c)
[thermostat demo](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/thermostat/thermostat.c)
[sensor demo](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/temperature_sensor/temperature_sensor.c)
Each of them doing a specific http call.

Since I am only familiar with the esp2866-arduino coding and I am really not a programmer, I think I'm gonna need help here...

RavenSystem, 2018-05-16:
Since this is a esp-open-rtos based project, maybe you want to use this instead: https://github.com/SuperHouse/esp-open-rtos/tree/master/examples/http_get

supersjimmie, 2018-05-17:
To start, I have created a copy of the led example and renamed led to switch in the files.
Then I replaced the on/off function, like:
```
void switch_write(bool on) {
	if(on) {
		http_get_task("?switch=1");
	} else {
		http_get_task("?switch=0");
	}
}
```
I added the http_get function from @RavenSystem  and slightly modified it to simply run once per call (not a while(1) loop), and added ```char *cmd``` as parameter.
In a new http_get.h file, I defined server, port and path.

So, hopefully, this would then create a homekit demo with a switch that calls my own API with either 1 or zero for the switch.

Next is adding the /API?switch=x to my other esp8266 code and toggle the real switch (which is controlled over a 433MHz transmitter).

supersjimmie, 2018-05-17:
No luck on this attempt.
The esp boots and I can add it on my iPhone as "Switch".
But then the esp bails out, with this logging:
```
HTTP get task starting...
Running DNS lookup for 192.168.0.219...
DNS lookup succeeded. IP=192.168.0.219
... allocated socket
Task stack overflow (high water mark=0 name="switch identify")
... socket connect failed.
Task stack overflow (high water mark=0 name="switch identify")
>>> homekit_serv
ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

supersjimmie, 2018-05-17:
So after some debugging and stripping the obsolete code, I've got the basics working.
I have a button on my iPhone and when I click it, the API call goes to the primary esp8266 and toggles the 433MHz switch.

Next will be finding out how to add more services and/or accessories.
(dimmer with 3 push buttons, reading temperature, ...)

supersjimmie, 2018-05-19:
Got this basically working, this can be closed. 👍

closed 2018-05-19 by supersjimmie