maximkulkin/esp-homekit-demo #77: Undefined references in mdnsresponder.c

q2222, 2018-05-11:
Are there additional steps needed for the ESP8266 since the support of the ESP32 was integrated?- I can successfully compile and generate the *.bin files for the ESP32 (thank you very much for your great work!), however I can no longer compile my ESP8266 projects and not even the pure examples/led project. I get the following errors:

> CC /Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c
/Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c: In function 'mdns_send_mcast':
/Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c:689:9: warning: implicit declaration of function 'LOCK_TCPIP_CORE' [-Wimplicit-function-declaration]
         LOCK_TCPIP_CORE();
         ^
/Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c:691:9: warning: implicit declaration of function 'UNLOCK_TCPIP_CORE' [-Wimplicit-function-declaration]
         UNLOCK_TCPIP_CORE();
         ^
...
...
...
...
...
...
...
Removing unwanted objects from /Volumes/ESP/esp-open-rtos/lib/libgcc.a
Removing unwanted objects from /Volumes/ESP/esp-open-rtos/libc/xtensa-lx106-elf/lib/libc.a
SDK processing stage 1: Removing unwanted objects from /Volumes/ESP/esp-open-rtos/lib/libmain.a
SDK processing stage 2: Renaming symbols in SDK library build/sdklib/libmain_stage1.a -> build/sdklib/libmain.a
SDK processing stage 1: Removing unwanted objects from /Volumes/ESP/esp-open-rtos/lib/libnet80211.a
SDK processing stage 2: Renaming symbols in SDK library build/sdklib/libnet80211_stage1.a -> build/sdklib/libnet80211.a
SDK processing stage 1: Removing unwanted objects from /Volumes/ESP/esp-open-rtos/lib/libphy.a
SDK processing stage 2: Renaming symbols in SDK library build/sdklib/libphy_stage1.a -> build/sdklib/libphy.a
SDK processing stage 1: Removing unwanted objects from /Volumes/ESP/esp-open-rtos/lib/libpp.a
SDK processing stage 2: Renaming symbols in SDK library build/sdklib/libpp_stage1.a -> build/sdklib/libpp.a
SDK processing stage 1: Removing unwanted objects from /Volumes/ESP/esp-open-rtos/lib/libwpa.a
SDK processing stage 2: Renaming symbols in SDK library build/sdklib/libwpa_stage1.a -> build/sdklib/libwpa.a
LD build/led.out
./build/homekit.a(mdnsresponder.o): In function `mdns_add_to_answer':
/Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c:537: undefined reference to `LOCK_TCPIP_CORE'
/Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c:537: undefined reference to `UNLOCK_TCPIP_CORE'
./build/homekit.a(mdnsresponder.o): In function `mdns_send_mcast':
/Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c:537: undefined reference to `LOCK_TCPIP_CORE'
/Volumes/ESP/esp-homekit-demo/components/homekit/src/mdnsresponder.c:537: undefined reference to `UNLOCK_TCPIP_CORE'
collect2: error: ld returned 1 exit status
make: *** [build/led.out] Error 1

Could you give me a hint, please what I am doing wrong?

maccoylton, 2018-05-11:
Works ok form me, think that comes for the lip library, therefore check your version of esp-open-rtos

q2222, 2018-05-11:
Thank you very much for the hint!- Everything OK again.

closed 2018-05-11 by q2222