maximkulkin/esp-homekit-demo #79: WiFi Portal on LED Strip example?

idumzaes, 2018-05-15:
Is it possible to have the LED Strip example fall back to the WiFi portal if it cannot connect to the designated WiFi network?
I’ve been trying to get it to but have been unable to. 
Thanks.

maximkulkin, 2018-05-19:
Code please

RavenSystem, 2018-05-20:
You must implement that functionality by yourself, for example, using https://github.com/maximkulkin/esp-wifi-config or another lib that you want.

maximkulkin, 2018-05-20:
@idumzaes As name mentions, the code provided in this repository are examples, which means they are not meant to cover everybody's needs or evolve in a assembly kit with different options. The led_strip example uses hard coded WiFi credentials. There is an example of how to do a captive portal style WiFi configuration: take a look at wifi_config or sonoff_basic examples on how to use wifi_config library for that.

idumzaes, 2018-05-20:
It looks like the magic_home_strip example may have what I'm looking for. I'll compile that and try it out.
Thank you guys for your responses.

closed 2018-06-07 by idumzaes