maximkulkin/esp-homekit-demo #85: Sample of dynamically generated services

medmile, 2018-05-22:
I am trying to create dynamically generated services:

`homekit_accessory_t *accessories[2];

homekit_service_t *services[3];

homekit_server_config_t config;

void config_init() {
    services[0] = HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
        HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
        //&accessory_name,
        HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
        HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
        HOMEKIT_CHARACTERISTIC(MODEL, "MyLED"),
        HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
        HOMEKIT_CHARACTERISTIC(IDENTIFY, controller_identify),
        NULL
    });
    
    services[1] = HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
        HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
        &led_on,
        NULL
    });
    
    services[2] = NULL;
    
    accessories[0] = HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=services);
    
    accessories[1] = NULL;
    
    config.accessories = accessories;
    config.password = "111-11-111";
    
    homekit_server_init(&config);
}`

But it don't work.

Can you help me?

supersjimmie, 2018-05-22:
I think your compiler fails on this:
```
	for (int i = 0; to 3; i++){
```
Try a for loop like this:
```
	for (int i = 0; i<3; i++){
```

maximkulkin, 2018-05-23:
I take it back, you can do it easier:
```c
#include <string.h>

void *memdup(const void *src, size_t size) {
    void *dst = malloc(size);
    return dst ? memcpy(dst, src, size) : NULL;
}

#define NEW_CHARACTERISTIC(name, ...) \
    (homekit_characteristic_t*)memdup((homekit_characteristic_t[]){HOMEKIT_CHARACTERISTIC_(name, ##__VA_ARGS__)}, sizeof(homekit_characteristic_t))

homekit_accessory_t **accessories = calloc(2, sizeof(homekit_accessory_t*));

homekit_accessory_t *lightbulb = accessories[0] = calloc(1, sizeof(homekit_accessory_t));
lightbulb->id = 1;
lightbulb->category = homekit_accessory_category_lightbulb;
lightbulb->services = calloc(3, sizeof(homekit_service_t*));

homekit_service_t *lightbulb_info = accessories accessories[0].services[0] = calloc(1, sizeof(homekit_service_t));
lightbulb_info->type = HOMEKIT_SERVICE_ACCESSORY_INFO;
lightbulb_info->characteristics = calloc(7, sizeof(homekit_characteristic_t*));

lightbulb_info->characteristics[0] = NEW_HOMEKIT_CHARACTERISTIC(NAME, "My accessory");
lightbulb_info->characteristics[1] = NEW_HOMEKIT_CHARACTERISTIC(MANUFACTURER, "medmile");
// ...
```

You probably can compact it even more.

medmile, 2018-05-29:
I am trying to do this, bun it do not work. What is wrong?

`/*
 * Example of using esp-wifi-config library to join
 * accessories to WiFi networks.
 *
 * When accessory starts without WiFi config or
 * configure WiFi network is not available, it creates
 * it's own WiFi AP "my-accessory-XXXXXX" (where XXXXXX
 * is last 3 digits of accessory's mac address in HEX).
 * If you join that network, you will be presented with
 * a page to choose WiFi network to connect accessory to.
 *
 * After successful connection accessory shuts down AP.
 *
 */

#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_system.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <etstimer.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"
#include <wifi_config.h>

const int led_gpio = 2;

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void led_on_callback(homekit_characteristic_t *_ch, homekit_value_t on, void *context);

homekit_characteristic_t led_on = HOMEKIT_CHARACTERISTIC_(
    ON, false, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(led_on_callback)
);

void controller_init() {
    gpio_enable(led_gpio, GPIO_OUTPUT);
    led_write(led_on.value.bool_value);
}

void led_on_callback(homekit_characteristic_t *_ch, homekit_value_t on, void *context) {
    led_write(led_on.value.bool_value);
}

void controller_identify_task(void *_args) {
    for (int i=0; i<3; i++) {
        for (int j=0; j<2; j++) {
            led_write(true);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            led_write(false);
            vTaskDelay(100 / portTICK_PERIOD_MS);
        }

        vTaskDelay(250 / portTICK_PERIOD_MS);
    }

    led_write(led_on.value.bool_value);

    vTaskDelete(NULL);
}

void controller_identify(homekit_value_t _value) {
    printf("LED identify\n");
    xTaskCreate(controller_identify_task, "Controller identify", 128, NULL, 2, NULL);
}

homekit_server_config_t config;

void *memdup(const void *src, size_t size) {
    void *dst = malloc(size);
    return dst ? memcpy(dst, src, size) : NULL;
}

#define NEW_HOMEKIT_CHARACTERISTIC(name, ...) \
(homekit_characteristic_t*)memdup((homekit_characteristic_t[]){HOMEKIT_CHARACTERISTIC_(name, ##__VA_ARGS__)}, sizeof(homekit_characteristic_t))

void config_init() {
    
    homekit_accessory_t **accessories = calloc(2, sizeof(homekit_accessory_t*));
    
    homekit_accessory_t *contoroller = accessories[0] = calloc(1, sizeof(homekit_accessory_t));
    contoroller->id = 1;
    contoroller->category = homekit_accessory_category_lightbulb;
    contoroller->services = calloc(3, sizeof(homekit_service_t*));
    
    homekit_service_t *contoroller_info = contoroller->services[0] = calloc(1, sizeof(homekit_service_t));
    contoroller_info->type = HOMEKIT_SERVICE_ACCESSORY_INFORMATION;
    contoroller_info->characteristics = calloc(7, sizeof(homekit_characteristic_t*));
    
    contoroller_info->characteristics[0] = NEW_HOMEKIT_CHARACTERISTIC(NAME, "Controller");
    contoroller_info->characteristics[1] = NEW_HOMEKIT_CHARACTERISTIC(MANUFACTURER, "medmike");
    contoroller_info->characteristics[2] = NEW_HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D");
    contoroller_info->characteristics[3] = NEW_HOMEKIT_CHARACTERISTIC(MODEL, "MyLED");
    contoroller_info->characteristics[4] = NEW_HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1");
    contoroller_info->characteristics[5] = NEW_HOMEKIT_CHARACTERISTIC(IDENTIFY, controller_identify);
    contoroller_info->characteristics[6] = NULL;
    
    homekit_service_t *switcher = contoroller->services[1] = calloc(1, sizeof(homekit_service_t));
    switcher->type = HOMEKIT_SERVICE_LIGHTBULB;
    switcher->characteristics = calloc(3, sizeof(homekit_characteristic_t*));
    
    switcher->characteristics[0] = NEW_HOMEKIT_CHARACTERISTIC(NAME, "Sample LED");
    switcher->characteristics[1] = &led_on;
    switcher->characteristics[2] = NULL;
    
    contoroller->services[2] = NULL;
    
    accessories[1] = NULL;
    
    config.accessories = accessories;
    config.password = "111-11-111";
    
    homekit_server_init(&config);
}

void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };
    
    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
    
    config_init();
}

void user_init(void) {
    
    uart_set_baud(0, 115200);

    wifi_init();
    
    controller_init();
}
`

fanlessfan, 2018-09-10:
@maximkulkin Could you add an example for dynamically generated services? for example using ds18b20 temperature sensor. the services will depend on how many sensors attached.

thx

maximkulkin, 2019-01-27:
@fanlessfan Check out the code above

maximkulkin, 2019-09-04:
Example is called dynamic_services.

closed 2019-09-04 by maximkulkin