maximkulkin/esp-homekit-demo #86: Interrupts on GPIO12 and GPIO14 do not appear to work

bwjohns4, 2018-05-24:
In the sonoff_basic example, I’ve tried to add two more buttons using GPIO12 and GPIO14. There are no errors, and button1 still works. I added print statements to the callbacks, just to see if they were even called when pin changes. Nothing happens on those two, but the GPIO 0 button still works fine, and does print a verification message during the ISR. Does the Homekit portion or other libraries perhaps use pins 12 and 14 differently and maybe overriding what I’m trying to do? Do those two pins perhaps not support that type of interrupt? I don’t know what else to check.

void user_init(void) {
    uart_set_baud(0, 115200);

    create_accessory_name();

     wifi_config_init("sonoff-switch", NULL, on_wifi_ready);
    gpio_init();

    if (button_create(button1_gpio, 0, 4000, button1_callback)) {
        printf("Failed to initialize button\n");
    }
    if (button_create(button2_gpio, 0, 4000, button2_callback)) {
        printf("Failed to initialize button\n");
    }
    if (button_create(button3_gpio, 0, 4000, button3_callback)) {
        printf("Failed to initialize button\n");
    }
}

supersjimmie, 2018-05-25:
I can't find the call that sets the button-pin as input, like ```gpio_enable(button->gpio_num, GPIO_INPUT);```.
That could explain why they don't work (but does not explain why one works :) ).

RavenSystem, 2018-05-25:
GPIO-0 doesn't need initialization; the others, yes.

korstiaanS, 2018-05-28:
As you already know, GPIO12 and GPIO14 are not necessary on PIN 12 and 14 (as you mention in your text).
FYI; I already used GPIO12 as an interrupt input without any problem.

konifer44, 2018-08-06:
https://www.esp32.com/viewtopic.php?t=2687 it's also the same situation if You want use as Input

closed 2019-01-27 by maximkulkin