maximkulkin/esp-homekit-demo #88: Wifi persistency

neptunix, 2018-05-27:
Hi! Thanks for a great project. I've got a question about wifi persistency.

I've flashed sonoff basic, specifying wifi pass in wifi.h

It started own wifi AP after flashing (didn't connect automatically to the supplied data in wifi.h)
Is it because `$(abspath ../../components/wifi_config)` in Makefile makes it ignore the setting?

Next.
I've connected it to an existing network and was able to register as a HomeKit device.
Then I disconnected the device from usb and connected to the AC. Device has booted and worked fine for a little while (~15-20 min).
Then HomeKit does not connect, device is pingable from main network
```
64 bytes from 10.3.1.139: icmp_seq=901 ttl=255 time=22.983 ms
64 bytes from 10.3.1.139: icmp_seq=902 ttl=255 time=15.759 ms
```
and it announces wifi AP at the same time!

I reconnected AC, provided credentials again, added as HomeKit and everything repeats.

So it's a loop:
* device is on main wifi network (pingable, HomeKit not advertised) and advertises its AP at the same time
* provide credentials - AP advertisement stops
* HomeKit is advertised and registers on iOS

neptunix, 2018-05-27:
Update:
I decided not to connect device to HomeKit network to test. It's 4 hours and no failures so far: stays on the wifi network, 100% time pingable and HomeKit is advertised.

neptunix, 2018-05-29:
Hi Rens, @renssies!

Is there any way that HomeKit press can interfere with button_callback?
Looking at the code I don't see where it does, but it looks like it really does.

Seems like relay_gpio somehow triggers button_callback...

renssies, 2018-05-29:
It might be that when the sonoff resets (because of an error for instance) that GPIO0 is still pulled low. Causing the Sonoff to reset. 

You could test it by only allowing the sonoff to be reset (so the callback to be handled) if the wifi has been configured. You could do this with a simple boolean you set on wifi_ready and check in the button callback.

neptunix, 2018-05-29:
I've removed original issue text, as it was my misunderstanding of a problem.

The device currently connects to a wifi and HomeKit properly. But triggering On/Off state from HomeKit callback (not pressing on GPIO0) leads to a reset.

Yes, it makes sense in checking if wifi has been configured in button callback. But it is not related to a problem why the devices makes a reset.

renssies, 2018-05-29:
With a reset in this case I'm meant a "soft reset" which basically restarts the device. 

I would try the approach that resetting using the button is only allowed after wifi is ready, just to see if it helps

kiralikbeyin, 2018-06-20:
Are you trying sonoff with usb power or 220V ? Try 220V if you are using usb..

peros550, 2018-06-20:
I am using 220V power

RavenSystem, 2018-06-20:
Similar issue: https://github.com/RavenSystem/esp-homekit-devices/issues/12

streicherlouw, 2018-07-16:
I am experiencing this issue as well. It seems that it may the the result of spurious interrupts from electrical noise on GPIO0, probably caused by EMF from either the relay's own switching or the load current being switched through the relay. I am investigating how one may design the interrupt handler in button.c to better handle super short pulses on GPIO0.

kiralikbeyin, 2018-07-16:
Experince

I was using delay when relay is on other relay is off (only one relay is on at the same time)
Removed ```  vTaskDelay(250 / portTICK_PERIOD_MS); ``` no reset now..

streicherlouw, 2018-07-21:
In my case, the problem seem to have been caused by extremely short "presses of the button" caused not by actual button presses, but by noise on the interrupt line caused by the relay's switching.  I was able to fix the problem by using the variable "button->last_press_time" not only as a timer, but also as a semaphore to indicate whether a valid press had been detected before before the release. In this code, button->last_press_time gets value=now only when a stable press is detected, otherwise it is set to 0, to indicate that a button release without a proceeding valid button press should be ignored. For this to work, it is also critical to initiate the button->last_press_time variable as 0, and not as "now".

Here is the contents of my working button.c 

`
#include <string.h>
#include <esplibs/libmain.h>
#include "button.h"

typedef struct _button {
    uint8_t gpio_num;
    button_callback_fn callback;

    uint16_t debounce_time;
    uint16_t long_press_time;

    bool pressed_value;

    uint32_t last_press_time;
    uint32_t last_event_time;

    struct _button *next;
} button_t;

button_t *buttons = NULL;

static button_t *button_find_by_gpio(const uint8_t gpio_num) {
    button_t *button = buttons;
    while (button && button->gpio_num != gpio_num)
        button = button->next;

    return button;
}

void button_intr_callback(uint8_t gpio) {
    button_t *button = button_find_by_gpio(gpio);
    if (!button)
        return;

    uint32_t now = xTaskGetTickCountFromISR();
    if ((now - button->last_event_time)*portTICK_PERIOD_MS < button->debounce_time) {
        // debounce time, ignore events
        button->last_press_time = 0;
        return;
    }
    button->last_event_time = now;
    if (gpio_read(button->gpio_num) == button->pressed_value) {
        // Record when the button is pressed down.
        button->last_press_time = now;
    } else {
        // The button is released. Handle the use cases.
        if ((now - button->last_press_time) * portTICK_PERIOD_MS > button->long_press_time) {
         	if (button->last_press_time > 0)
            	{
            	button->callback(button->gpio_num, button_event_long_press);
            	button->last_press_time = 0;
            	}
        } else {
            if (button->last_press_time > 0)
            	{
            	button->callback(button->gpio_num, button_event_single_press);
            	button->last_press_time = 0;
            	}
        }
    }
}

int button_create(const uint8_t gpio_num, bool pressed_value, uint16_t long_press_time, button_callback_fn callback) {
    button_t *button = button_find_by_gpio(gpio_num);
    if (button)
        return -1;

    button = malloc(sizeof(button_t));
    memset(button, 0, sizeof(*button));
    button->gpio_num = gpio_num;
    button->callback = callback;
    button->pressed_value = pressed_value;

    // times in milliseconds
    button->debounce_time = 50;
    button->long_press_time = long_press_time;

    uint32_t now = xTaskGetTickCountFromISR();
    button->last_event_time = now;
    button->last_press_time = 0;

    button->next = buttons;
    buttons = button;

    gpio_set_pullup(button->gpio_num, true, true);
    gpio_set_interrupt(button->gpio_num, GPIO_INTTYPE_EDGE_ANY, button_intr_callback);

    return 0;
}

void button_delete(const uint8_t gpio_num) {
    if (!buttons)
        return;

    button_t *button = NULL;
    if (buttons->gpio_num == gpio_num) {
        button = buttons;
        buttons = buttons->next;
    } else {
        button_t *b = buttons;
        while (b->next) {
            if (b->next->gpio_num == gpio_num) {
                button = b->next;
                b->next = b->next->next;
                break;
            }
        }
    }

    if (button) {
        gpio_set_interrupt(gpio_num, GPIO_INTTYPE_EDGE_ANY, NULL);
    }
}

`

neptunix, 2018-07-25:
I've too tested the code and its reliable now. No more resets! Thanks

peros550, 2018-07-31:
The above proposed code (button.c) seems to solve the issue for me too. I am going to monitor behavior for the next days and report back. 

Thanks.

LizardM4, 2018-11-02:
I am also experiencing the same problem. Rapid on/off switching will cause the device to go back to the initial state (Wifi configuration). I tried the `button.c` provided by @streicherlouw and the problem is gone. Any plan on merging this?

PR #110 (2019-03-11): Button trembling fix
Resolves #88 

Code provided by @streicherlouw
closed 2019-03-11 by maximkulkin