maximkulkin/esp-homekit-demo #89: Latest version

korstiaanS, 2018-05-28:
Hi,

I'm not not an Git expert...
I tried to get latest version of all modules by doing:

`git clone https://github.com/maximkulkin/esp-homekit-demo.git`
followed by
`git submodule update --init --recursive`

But if I look afterwards I see that homekit has not the latest version.

`Submodule path 'components/homekit': checked out 'a592dc3587fb3b5b2f27d3ed068186cd6ec8d677'`

And in a graphical tool I see:

[image]

How do I get the latest version?

neptunix, 2018-05-28:
Right click on the top line -> Checkout

korstiaanS, 2018-05-29:
Hi,

Yes that I know, but that could not be the way to work to get the latest version isn't it? If you follow the readme and execute some terminal commands you think you have the latest versions but that is not the case. What is the reason that the pointer is not pointed to the latest?

peros550, 2018-06-16:
If there is no graphical tool, what is the correct command to get the latest version?

maximkulkin, 2019-04-04:
Since Homekit is a submodule here, it always references particular commit and you have to update it manually. The command for that is:
```
cd components/common/homekit
git pull origin master
```

closed 2019-04-04 by maximkulkin