maximkulkin/esp-homekit-demo #91: Unable to add CURRENT_AMBIENT_LIGHT_LEVEL

supersjimmie, 2018-05-29:
I can add multiple different sensors to several accessories, but when I add a sensor for ambient light level the pairing keeps failing.
I tried the sensor as part of an accessory type homekit_accessory_category_sensor, homekit_accessory_category_other, homekit_accessory_category_lightbulb but none of them makes any difference.   When I leave all code and just change this to a humidity_sensor the pairing works fine.

What type of accessory_category is needed for the ambient_light_level?
Or what else could be the reason?

supersjimmie, 2018-05-31:
In file types.h, I can find
```
    homekit_accessory_category_lightbulb = 5,
...
    homekit_accessory_category_sensor = 10,
...
```
But there is no light_sensor.
When searching with google, I guess the ```CURRENT_AMBIENT_LIGHT_LEVEL``` is a characteristic for a service ```HOMEKIT_SERVICE_LIGHT_SENSOR``` which is part of some kind of ```homekit_accessory_category_light_sensor```  or ```homekit_accessory_category_lightsensor```?

supersjimmie, 2018-05-31:
Found the problem, I was setting the inital value to 0 (just like with many other devices), but the lux cannot be 0.

closed 2018-05-31 by supersjimmie

Korcsog, 2018-06-03:
@supersjimmie The accessory type is `homekit_accessory_category_sensor`, and the value of lux could be 0, if you set minValue to 0 too! I'm using this with minValue 0, and still works properly!

Mine:

`HOMEKIT_CHARACTERISTIC_(CURRENT_AMBIENT_LIGHT_LEVEL, 0, .min_value = (float[]) {0},);`